FastLED/FastLED #1877: Documentation references MIDI header file

mkeveney, 2025-01-29:
Documentation Topics include an item called "Input" which documents part of the MIDI interface.

https://fastled.io/docs/topics.html
https://fastled.io/docs/group__input.html

I suspect this was created is due to the presence of the "LuminescentGrand" example; the only place I find references to MIDI in the repository.

zackees, 2025-01-29:
Midi will eventually go into the core lib so not going to remove

mkeveney, 2025-01-29:
?
You mean the "Arduino" core?
If you mean the FastLED core, Midi seems like an odd thing to include.  What's the rationale?

zackees, 2025-01-29:
I mean it's going to go into the FastLED ~core~ sensor lib.

> What's the rationale?

FastLED is getting app features like the fx and sensors library. For why we'd want this, see our unannounced web-compiler:

https://github.com/zackees/fastled-wasm

In order to make it easy for designers to use, more stuff has to be included in FastLED. For example a midi library in FastLED means the browser can connect to it and act as a passthrough to web-midi and people can test their complicated visual sketches with 3 second compile times and 1 second deploys to the browser. This enables live coding.

If you haven't seen it yet, i recommend trying out the new web compiler. It's next level stuff.

closed 2025-01-29 by zackees