FastLED/FastLED #1878: XIAO ESP32C3 not working with FastLED

jcabdi, 2025-01-29:
### CODE:
`
#include <Arduino.h>
#include <FastLED.h>
#include <rom/rtc.h>
#include "esp_chip_info.h"

#define NUM_LEDS 15
#define LED_PIN 3

CRGB leds[NUM_LEDS];

void setup() {
    Serial.begin(57600);
    Serial.println("- Starting up...");
    esp_reset_reason_t reason = esp_reset_reason();
    Serial.print("Reset reason: ");
    Serial.println((int)reason);
    Serial.printf("ESP32:\n");
    Serial.printf("  model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
    Serial.printf("  cores: %d\n", ESP.getChipCores());
    Serial.printf("  mac: %d\n", ESP.getEfuseMac());
    Serial.println();
    FastLED.addLeds<WS2812B, LED_PIN>(leds, NUM_LEDS);
    FastLED.setBrightness(200);
  }

void loop() {
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Red;
    Serial.println(i);
    FastLED.show();
    delay(100);
  }
}
`
### The chip is rebooting itself and below is the monitor capture:

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403821a4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
E (77) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='7985abd8'
Image checksum='19dfc1ee'
- Starting up...
Reset reason: 4
ESP32:
  model = ESP32-C3 Rev 4
  cores: 1
  mac: -1086596372

0
Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x00000004  RA      : 0x403801ee  SP      : 0x3fc8e010  GP      : 0x3fc8c600  
TP      : 0x3fc897dc  T0      : 0x7fffffff  T1      : 0x00000005  T2      : 0x00000005  
S0/FP   : 0x3fc93020  S1      : 0x8000000a  A0      : 0x3fc93020  A1      : 0x3fc8e02c  
A2      : 0x3fc93018  A3      : 0x00000004  A4      : 0x600c2000  A5      : 0x00000050  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x00001881  S3      : 0x00000004  
S4      : 0x3fc8eb5c  S5      : 0x3fc8f000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x3fc8f538  T4      : 0x0000000c  T5      : 0x0000000d  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000001  MTVAL   : 0x00000004  
MHARTID : 0x00000000  

Stack memory:
3fc8e010: 0x40387e62 0x40387ea6 0x3fc8e090 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e030: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881
3fc8e050: 0x00000000 0x00000000 0x00001881 0x00000004 0x3fc8eb5c 0x3fc8f000 0x00000000 0x00000000
3fc8e070: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f538 0x0000000b 0x0000000c 0x00000000
3fc8e090: 0x00000000 0x00000000 0x00000000 0x403803f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc8e0b0: 0x00000000 0x3fc8f000 0x3fc8eb5c 0x00000004 0x00001881 0x8000000a 0x3fc93020 0x403801ee
3fc8e0d0: 0x40387e62 0x40387ea6 0x3fc8e150 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e0f0: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881
3fc8e110: 0x00000000 0x00000000 0x00001881 0x00000004 0x3fc8eb5c 0x3fc8f000 0x00000000 0x00000000
3fc8e130: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f538 0x0000000a 0x0000000b 0x00000000
3fc8e150: 0x00000000 0x00000000 0x00000000 0x403803f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc8e170: 0x00000000 0x3fc8f000 0x3fc8eb5c 0x00000004 0x00001881 0x8000000a 0x3fc93020 0x403801ee
3fc8e190: 0x40387e62 0x40387ea6 0x3fc8e210 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e1b0: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881
3fc8e1d0: 0x00000000 0x00000000 0x00001881 0x00000004 0x3fc8eb5c 0x3fc8f000 0x00000000 0x00000000
3fc8e1f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f538 0x00000009 0x0000000a 0x00000000
3fc8e210: 0x00000000 0x00000000 0x00000000 0x403803f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc8e230: 0x00000000 0x3fc8f000 0x3fc8eb5c 0x00000004 0x00001881 0x8000000a 0x3fc93020 0x403801ee
3fc8e250: 0x40387e62 0x40387ea6 0x3fc8e2d0 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e270: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881
3fc8e290: 0x00000000 0x00000000 0x00001881 0x00000004 0x3fc8eb5c 0x3fc8f000 0x00000000 0x00000000
3fc8e2b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f538 0x00000008 0x00000009 0x00000000
3fc8e2d0: 0x00000000 0x00000000 0x00000000 0x403803f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc8e2f0: 0x00000000 0x3fc8f000 0x3fc8eb5c 0x00000004 0x00001881 0x8000000a 0x3fc93020 0x403801ee
3fc8e310: 0x40387e62 0x40387ea6 0x3fc8e390 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e330: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881
3fc8e350: 0x00000000 0x00000000 0x00001881 0x00000004 0x3fc8eb5c 0x3fc8f000 0x00000000 0x00000000
3fc8e370: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f538 0x00000007 0x00000008 0x00000000
3fc8e390: 0x00000000 0x00000000 0x00000000 0x403803f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc8e3b0: 0x00000000 0x3fc8f000 0x3fc8eb5c 0x00000004 0x00001881 0x8000000a 0x3fc93020 0x403801ee
3fc8e3d0: 0x40387e62 0x40387ea6 0x3fc8e450 0x3fc8c600 0x3fc897dc 0x7fffffff 0x403803de 0x00000005
3fc8e3f0: 0x3fc93020 0x60016000 0x00000005 0x3fc8e02c 0x3fc93018 0x00000004 0x600c2000 0x00001881

ELF file SHA256: be2d373708daf8d3

Its is happening with older versions also.

sutaburosu, 2025-01-29:
>  E (77) esp_core_dump_flash: Core dump data check failed:
> Calculated checksum='7985abd8'
> Image checksum='19dfc1ee'

Your device is rebooting because it has detected that the contents of the Flash RAM seems to be corrupted. This may be a problem with the upload procedure, or it may be the Flash chip in the ESP has failed.

Either way, using the same code on a different, but identical, device should give success.

jcabdi, 2025-01-30:
Thanks for your fast response. Actually I found the problem, I uploaded the sketch with the Arduino IDE and everything worked find, then I realized that the problem was that the platformio was outdated, one I update platformio, the problem was solved.
Regards.

sutaburosu, 2025-01-30:
Great news that you resolved the problem. Can you close this issue now please?

closed 2025-01-30 by jcabdi