FastLED/FastLED #1880: is there an overview of the objects (classes) with properties/methods available

dbyy, 2025-01-31:
As described in the title, is there a documentation as a class overview with all properties (public/hidden) and the methods pro class available?

I was reading the wiki but I didn't find an overview as it is 'usually' available in OO programming.
:-)

sutaburosu, 2025-01-31:
There is a link on [fastled.io](https://fastled.io/) which leads to [the API documentation](https://fastled.io/docs/).

I'm wondering where in the wiki would be best place to put a link to make this more discoverable. Were there specific areas of the Wiki that you searched?

dbyy, 2025-02-01:
- The best place would be perhaps Wiki -> References
- I went to (almost) through all entries in the wiki (to learn FastLED) LOL
- To give you an impression what I was visioning, I attach an example I quickly created as a draft (done in Treeline). In the rush I couldn't find a good example on the web to copy it. But I know they exist. I used them in the past during my active time as developer.
**This example absolutely not correct and not complete.**

[image]

closed 2025-03-13 by zackees