FastLED/FastLED #1881: LC8816E Support request

notsolowki, 2025-02-01:
I ordered some LC8816E from alibaba and tried to use them with the WS2815 config in fastled but i dont think its working right. forgive my knowledge but white appears to work but when i set to another color it seems like each led is a different color. 

How can i help make this chips timings compatible with fastled?

zackees, 2025-03-11:
Can you provide a data sheet?

We had someone mention this chipset 3 months ago on our reddit board:

https://www.reddit.com/r/FastLED/comments/1h10h57/lc8816e_support/

My best advice right now is to use WS2812 or WS2815 (which you are already doing) and see if one of those works. If that doesn't work then use the `#define FASTLED_OVERCLOCK float` setting to compress/expand the clock timings. Typically these chipsets are only strict on TH1 (time for high bit) setting and the THL (time for low bit) is more relaxed.

If you find something that works then let me know and I'll add it to our led selection.

zackees, 2025-06-24:
> I ordered some LC8816E from alibaba and tried to use them with the WS2815 config in fastled but i dont think its working right. forgive my knowledge but white appears to work but when i set to [...]

Try using RGBW mode.

closed 2026-08-20 by zackees