FastLED/FastLED #1882: Crash when FASTLED_ESP32_USE_CLOCKLESS_SPI and WiFi enabled in 3.9.13

Johboh, 2025-02-02:
Hi!
I have been using PlatformIO and FastLED 3.6.0, and FastLED has been working well.
Now I have switched from PlatformIO in favor for using ESP-IDF and the arduino component + FastLED as a component.

It all works well up until I enable WiFi. Once enable, the led panel is glitching where some leds switch color (black to blue) and so on. There are a couple of github issues regarding this since arlier, and the suggested fix to set `FASTLED_ALLOW_INTERRUPTS = 0` and/or `FASTLED_INTERRUPT_RETRY_COUNT=0` does not help in my case.

So instead, I tried using the clockless SPI (`FASTLED_ESP32_USE_CLOCKLESS_SPI`). If I don't enable WiFi, it works without any flickering, but once WiFi is enabled, I get a crash instead, see below.

Code:
```c++
#include <Arduino.h>
#define FASTLED_ESP32_USE_CLOCKLESS_SPI
#include "credentials.h" // Defines user_hostname, user_wifi_ssid and user_wifi_password
#include <FastLED.h>
#include <cstring>
#include <driver/gpio.h>
#include <esp_log.h>
#include <esp_mac.h>
#include <esp_netif.h>
#include <esp_system.h>
#include <esp_wifi.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <lwip/sys.h>
#include <nvs.h>
#include <nvs_flash.h>
#include <stdio.h>
#include <string>

#define TAG "s2-main"

#define PIN_LED GPIO_NUM_15
#define DATA_PIN GPIO_NUM_3
#define LED_TYPE WS2812B
#define COLOR_ORDER GRB
#define DEFAULT_BRIGHTNESS 96

const uint8_t kMatrixWidth = 21;
const uint8_t kMatrixHeight = 5;
#define NUM_LEDS (kMatrixWidth * kMatrixHeight)
CRGB _leds_current[NUM_LEDS];

void blinkAndSerialTask(void *pvParameters) {
  bool swap = false;
  while (1) {
    gpio_set_level(PIN_LED, swap);
    swap = !swap;
    ESP_LOGI(TAG, "Hello");
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}

void ledTask(void *pvParameters) {
  bool swap = false;
  while (1) {
    for (int i = 0; i < NUM_LEDS; i++) {
      _leds_current[i] = swap ? CRGB::Red : CRGB::Black;
      swap = !swap;
    }
    swap = false;

    FastLED.show();
    vTaskDelay(10 / portTICK_PERIOD_MS);
  }
}

void eventHandler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data) {
  // Log WiFi events
  if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
    ESP_LOGI(TAG, "WiFi disconnected");
[... 165 lines cut ...]
0x4008885c: CFastLED::show() at C:/stuff/sources/github/johboh/FastLED/src/FastLED.h:745
 (inlined by) ledTask(void*) at C:/stuff/sources/bitbucket/embedded/debug_or_testing_projects/s2_arduino_espidf/main/main.cpp:52
0x4002e65e: vPortTaskWrapper at C:/Users/johbo/esp/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

## Environment
ESP32-S2
ESP-IDF v5.3.2
FastLED: v3.9.13
espressif/arduino-esp32:  v3.1.1

## sdkconfig that differ from defaults (I tried without SPIRAM and same issue)
```
CONFIG_IDF_TARGET="esp32s2"
CONFIG_SPIRAM=y
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY=y
CONFIG_ESP_CONSOLE_USB_CDC=y
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=10
CONFIG_ESP_WIFI_RX_BA_WIN=6
CONFIG_FREERTOS_HZ=1000
CONFIG_LWIP_TCP_OOSEQ_MAX_PBUFS=4
```

zackees, 2025-02-02:
I'm sorry for that.

We use espressif's led_strip component for this. I'll file a bug with them.

zackees, 2025-02-02:
See:

https://github.com/espressif/idf-extra-components/issues/334

zackees, 2025-02-06:
The next release has some fixes for the older compiler toolchains and you'll now see a build badge to confirm it's working for older esp-idf build chains. Upgrade to the newest FastLED through manual download and install, then downgrade your other toolchains.