FastLED/FastLED #1884: FastLED with ESP8266 and mqtt.connect fails

nygma2004, 2025-02-05:
Please delete this if this is not an issue, but I was not sure where and how to ask.
I have a few year old project that I needed to recompile now. When I uploaded it to a D1 Mini, the code fails to connect to the MQTT server. The code seem to just hang and the WDT restarts the ESP. I tried a simple MQTT example code and it works fine.
This project uses a single neopixel for status and since it was not essential for the project, I commected out everything related to FastLED. And now MQTT works.
I am OK when it comes to creating functional code, but this is over my head.

This is my FastLED related code snippets:
```
#include <FastLED.h>

#define RGBLED_PIN D3        // Neopixel led D3
#define NUM_LEDS 1
#define LED_TYPE    WS2812
#define COLOR_ORDER GRB
#define BRIGHTNESS  32        // Default LED brightness.

CRGB leds[NUM_LEDS];

FastLED.addLeds<LED_TYPE, RGBLED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalSMD5050 );
FastLED.setBrightness( BRIGHTNESS );
leds[0] = CRGB::Pink;
FastLED.show();
```

Any pointers what and how to test would be appreciated.

I am using Arduino IDE 2.3.4, FastLED is updated to 3.9.13 now.

sutaburosu, 2025-02-05:
I have seen reports of many different problems with FastLED on ESP8266 using board support >= v3.0.0. Many of them can be worked around by rolling back the ESP8266 board support to v2.7.4. I wonder if your problem may also be in this category.

Obviously, this would not be a great solution in the long term, but it may allow you to make progress today.

nygma2004, 2025-02-06:
Most probably. I also have issues re-compiling some other projects for esp32 as well. To be honest I never considered moving back to the earlier version of the board support.
I re-installed 2.7.4 but I am getting complication errors now. I may have to roll back some of the libraries as well.

nygma2004, 2025-02-09:
I had a bit more time to test this. When I coded the sketch originally, I was using FastLED 3.3.3. I have downgraded to that version now and it is working. I am on these versions:

- Arduino IDE: 2.3.4
- ESP 8266 board library: 3.1.2
- FastLED: 3.3.3
- EspSoftwareSerial: 8.1.0
- Modbusmaster: 2.0.1

So the latest board manager library works with the 3.3.3 FastLED. I did not test at what version this broke.

zackees, 2025-03-03:
Fix is on it's way. Was already submitted.

zackees, 2025-03-05:
Fix was released in 3.9.14, it should go live shortly. Please re-oppen this bug if it's somehow not fixed.

closed 2025-03-05 by zackees