FastLED/FastLED #1885: HD108 support

Blaster1920, 2025-02-05:
Since the last release brought the HD107 to fastled, I've tried my best to write some code to port the HD108 as well, but I'm hitting a wall. I can see The pins are sending out data but it's not quite right yet, as the leds won't get set.
This is the barebone code that should make implementing the HD108 a little bit easier:

```
template <
    uint8_t DATA_PIN,
    uint8_t CLOCK_PIN,
    EOrder RGB_ORDER = RGB,
    uint32_t SPI_SPEED = DATA_RATE_MHZ(40)
>
class HD108Controller : public CPixelLEDController<RGB_ORDER> {  
    typedef SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED> SPI;
    SPI mSPI;

public:
    HD108Controller() {}

    virtual void init() override {
        mSPI.init();
    }

protected:
    virtual void showPixels(PixelController<RGB_ORDER> & pixels) override {
        mSPI.select();
        startFrame();
        while (pixels.has(1)) {
            uint8_t rGain = 1, gGain = 1, bGain = 1;  // Default gains to avoid LED flickering
            uint8_t r8, g8, b8;
            uint16_t r, g, b;

            // Corrected: Use FastLED's standard scaling function
            pixels.loadAndScaleRGB(&r8, &g8, &b8);

            // Scale from 8-bit (0-255) to 16-bit (0-65535) for HD108
            r = (uint16_t(r8) << 8) | r8;
            g = (uint16_t(g8) << 8) | g8;
            b = (uint16_t(b8) << 8) | b8;

            // Send data in the required format
            writeLedFrame(rGain, gGain, bGain, r, g, b);
            pixels.stepDithering();
            pixels.advanceData();
        }
        endFrame(pixels.size());
        mSPI.waitFully();
        mSPI.release();
    }

private:
    inline void startFrame() {
        mSPI.writeByte(0x00);
        mSPI.writeByte(0x00);
        mSPI.writeByte(0x00);
        mSPI.writeByte(0x00);
    }

    inline void endFrame(int nLeds) {
        int nBytes = (nLeds / 16) * 8;
        for (int i = 0; i < nBytes; i++) {
            mSPI.writeByte(0xFF);
        }
    }

inline void writeLedFrame(uint8_t rGain, uint8_t gGain, uint8_t bGain, uint16_t r, uint16_t g, uint16_t b) {
    uint8_t frame[8] = {
        uint8_t(0x80 | ((rGain & 0x1F) << 2) | ((gGain >> 3) & 0x03)),  // Gain Control (1st byte)
        uint8_t(((gGain & 0x07) << 5) | (bGain & 0x1F)),                // Gain Control (2nd byte)
        uint8_t(r >> 8), uint8_t(r & 0xFF),  // Red High & Low
        uint8_t(g >> 8), uint8_t(g & 0xFF),  // Green High & Low
        uint8_t(b >> 8), uint8_t(b & 0xFF)   // Blue High & Low
    };

    mSPI.writeBytes(frame, 8);
}

};

/// HD108HD - "high definition" bit-shift gamma controller
template <
    uint8_t DATA_PIN,
    uint8_t CLOCK_PIN,
    EOrder RGB_ORDER = GRB,  
    uint32_t SPI_SPEED = DATA_RATE_MHZ(40)
>
class HD108ControllerHD : public HD108Controller<
    DATA_PIN,
    CLOCK_PIN,
    RGB_ORDER,
    SPI_SPEED
> {
public:
    HD108ControllerHD() = default;
};
```

My personal implementation elsewhere worked well, albeit it's missing 16 bit depth. Note how I'm using 16MHz (probably can be increased) and spi mode 3.

```
// Initialize SPI
void HD108::begin() {
    spi_bus_config_t buscfg = {
        .mosi_io_num = _dataPin,
        .miso_io_num = -1,
        .sclk_io_num = _clockPin,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 64 * _numLEDs + 16 // Max transfer size
    };

    esp_err_t ret = spi_bus_initialize(_spiHost, &buscfg, SPI_DMA_CH_AUTO);
    assert(ret == ESP_OK);

    spi_device_interface_config_t devcfg = {
        .mode = 3,                   // SPI mode 3
        .clock_speed_hz = 15000000,  // 20 MHz clock speed
        .spics_io_num = -1,          // No CS pin
        .queue_size = 1,             // Transaction queue size
    };

    ret = spi_bus_add_device(_spiHost, &devcfg, &_spi);
    assert(ret == ESP_OK);

        sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(1, 0, 0, 0, 0, 0); // Turn off all LEDs
    }
    sendEndFrame();
}

// Send start frame
void HD108::sendStartFrame() {
    uint8_t startFrame[4] = {0x00, 0x00, 0x00, 0x00}; // 32-bit start frame
    spi_transaction_t t = {
        .length = 32, // 32 bits = 4 bytes
        .tx_buffer = startFrame,
    };
    spi_device_transmit(_spi, &t);
}

// Send LED frame
void HD108::sendLEDFrame(uint8_t redGain, uint8_t greenGain, uint8_t blueGain,
                         uint16_t red, uint16_t green, uint16_t blue) {
    uint8_t frame[8];

    // Automatically set gain to max if grayscale is 0, this avoids LEDs crashing
    if (red == 0) redGain = 1;
    if (green == 0) greenGain = 1;
    if (blue == 0) blueGain = 1;

    // Start Code + Current Gain Adjustment (15 bits)
    frame[0] = 0x80 | ((redGain & 0x1F) << 2) | ((greenGain >> 3) & 0x03); // Start code + Red MSB + Green MSB
    frame[1] = ((greenGain & 0x07) << 5) | (blueGain & 0x1F);              // Green LSB + Blue 5 bits

    // Encode RGB 16-bit values
    frame[2] = (red >> 8) & 0xFF;   // High byte of Red
    frame[3] = red & 0xFF;          // Low byte of Red
    frame[4] = (green >> 8) & 0xFF; // High byte of Green
    frame[5] = green & 0xFF;        // Low byte of Green
    frame[6] = (blue >> 8) & 0xFF;  // High byte of Blue
    frame[7] = blue & 0xFF;         // Low byte of Blue

    // Transmit the frame via SPI
    spi_transaction_t t = {
        .length = 64, // 64 bits = 8 bytes
        .tx_buffer = frame,
    };
    spi_device_transmit(_spi, &t);
}

// Send end frame
void HD108::sendEndFrame() {
    uint8_t endFrame[4] = {0xFF, 0xFF, 0xFF, 0xFF}; // At least 1 bit per LED
    spi_transaction_t t = {
        .length = (_numLEDs / 16) * 8, // End frame size
        .tx_buffer = endFrame,
    };
    spi_device_transmit(_spi, &t);
}

// Show function
void HD108::show(uint8_t* redGain, uint8_t* greenGain, uint8_t* blueGain,
                 uint16_t* red, uint16_t* green, uint16_t* blue) {
    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(
            redGain[i],    // Red gain
            greenGain[i],  // Green gain
            blueGain[i],   // Blue gain
            red[i],        // Red intensity
            green[i],      // Green intensity
            blue[i]        // Blue intensity
        );
    }
    sendEndFrame();
}

void HD108::clear() {
    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(0, 0, 0, 0, 0, 0); // Turn off all LEDs: gain and grayscale set to 0
    }
    sendEndFrame();
}

void HD108::showColour(const char* colorName, uint16_t ledIndex) {
    uint8_t redGain, greenGain, blueGain;
    uint16_t red, green, blue;

    getColourParameters(colorName, redGain, greenGain, blueGain, red, green, blue);

    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        if (i == ledIndex) {
            sendLEDFrame(redGain, greenGain, blueGain, red, green, blue);
        } else {
            sendLEDFrame(0, 0, 0, 0, 0, 0); // Turn off other LEDs
        }
    }
    sendEndFrame();
}
```

zackees, 2025-03-03:
Thanks for the demo, but if it doesn't work then that's a lot debugging.

I'll keep an eye out for this though and you'll get a credit on the contribution list if you can provide a working version.

Mpxxer, 2025-05-04:
Is there a way to speed up the hd108 implematation like contributiing or donate to a developer or something?

zackees, 2025-05-05:
> Is there a way to speed up the hd108 implematation like contributiing or donate to a developer or something?

Yes. I need some HD108 to test with.

The protocol is almost done via donation above, I suspect the reset code is off.

closed 2025-11-05 by zackees