FastLED/FastLED #1886: Flickering when using the ObjectFLED driver

lpaolini, 2025-02-09:
This issue is related to #1818.

One of my WS2812B LED strips, probably older than the other ones I have, does not work well when using the ObjectFLED driver.

I ran a few tests and this is the result:
- FastLED with [WS2812Serial driver](https://github.com/PaulStoffregen/WS2812Serial) => no problem
- FastLED with [ObjectFLED driver](https://github.com/KurtMF/ObjectFLED) => flickering noise
- Direct ObjectFLED (no FastLED) => no problem

@zackees Is FastLED initializing ObjectFLED with custom timings?

zackees, 2025-02-11:
> This issue is related to #1818.
>
> One of my WS2812B LED strips, probably older than the other ones I have, does not work well when using the ObjectFLED driver.
>
> I ran a few tests and this is the [...]

ObjectFLED has its own timings. try setting the overclock factor to 1.1 or 0.9, one if those should fix it.

This reason is the time for the on value needs adjusting. Overclocking or under kicking will make it slide into specification.

zackees, 2025-02-12:
[@KurtMF](https://github.com/KurtMF/)

I know we had talked about timings being passed from fastled to ObjectFLED before.

Do you have any advice on how I should proceed on this? Right now I'm telling users to fix the timing via overclocking hack.

lpaolini, 2025-02-13:
What is the reason for FastLED to customize ObjectFLED timings in the zero overclock case?
Also, do FastLED controllers map to individual ObjectFLED instances? If so, wouldn't it be possible to specify overclock when the controller is instantiated?
```
FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS, overclock_factor);
```