FastLED/FastLED #1888: Binary too large - build results on STM32F103

lgacnik, 2025-02-13:
I'm trying to find out about the expected binary overhead of the FastLED library. After compiling STM32 HAL and FastLED (without FX headers included), using GCC with -O2 flag I get the following binary size:

```
[build] Memory region         Used Size  Region Size  %age Used
[build]              RAM:        5240 B        20 KB     25.59%
[build]            FLASH:      122344 B        64 KB    186.68%
```

It's true that STM32F103 doesn't have extraordinary high amount of storage resources. Yet, apparently FlashLED (without FX headers) should be _lightweight_.

Can someone confirm whether what I'm seeing is expected build result or not? Maybe I can exclude certain, optional sources/headers from build in order to fit this library on the desired MCU?

lgacnik, 2025-02-15:
Resolved this issue by using `-ffunction-sections` and `-fdata-sections` as C build flags and `-Wl,--gc-sections` as linker flag. Together they exclude unused sections from resulting image.

The resulting binary size is now much less:

```
[build] Memory region         Used Size  Region Size  %age Used
[build]              RAM:        2648 B        20 KB     12.93%
[build]            FLASH:        9176 B        64 KB     14.00%
```

closed 2025-02-15 by lgacnik

zackees, 2025-03-03:
Note, use `-flto` for anything other than quick builds.