FastLED/FastLED #1890: Porting-related Issues and Proposed Improvements

lgacnik, 2025-02-16:
I'm trying to port the FastLED to STM32 platform. I understand I should modify headers under the `/src/platforms/arm/stm32/` folder. However the problem I see here is that relevant platform specific headers `fastpin_xxx.h` and `fastspi_xxx.h` are not included by core headers `fastspi.h` and `fastpin.h`. On the other hand the platform specific header `led_sysdefs_xxx.h` is included by the core header `led_sysdef.h` - why is the same not true for the `fastspi.h` and `fastpin.h`? **Am I supposed to include the non-included platform-specific headers myself in my build system globally?**

It would make sense for these platform-specific headers to be selectively included the same way as the core header `led_sysdef.h` handles includes.

If I would have to approach this matter as a FastLED contributor, e.g. for `fastpin.h`, I would selectively put selective platform-specific header selection from the start. If none of platform-specifc headers is included, use the `class FastPin` default definition you otherwise provide in `fastpin.h`. This would make it less troublesome to port. Now you first have to figure out that neither of platform specific headers that implement hardware capability are included and you _magically_ need to do that yourself.

Additionally, add some more general, wide-scope define that includes STM32 platform for any STM32 out there, not just STM32F1 and STM32F2. Leave the class implementation to user. I'm not saying you should provide compatibility for any STM32 family out there.

I'm heavily involved with STM32 so if further insights are needed regarding platform-specific stuff I can add my further input if needed.

lgacnik, 2025-02-16:
Additionally, the "[FastLED: Platform Porting Guide](https://fastled.io/docs/md__2home_2runner_2work_2_fast_l_e_d_2_fast_l_e_d_2_p_o_r_t_i_n_g.html)" doesn't cover how user should include platform-specific headers (e.g. for the `class FastLED` implementation). Would be nice if some of information asked here would be added there (in a couple of sentences).

zackees, 2025-03-03:
Hey, on the plus side, *it used to be worse*.

The best advice I have is to get into a compiled edit test cycle in fastled.

So for example if I want to build the `hy_tinystm103tb` project it would be this

```
git clone https://github.com/fastled/fastled
cd fastled
./compile hy_tinystm103tb --examples Blink
```

Everything will be installed automatically and it will then build the project.

The platform headers are decided by the define environment. Getting the environment can be tricky, but luckily everytime you invoke our `./compile` tool, I've had it write out a build_info.json file which will contain all of the build environment used to compile the project.

[image]

zackees, 2025-03-03:
Fyi here's the build_info.json for this platform above:

```json

{
    "hy_tinystm103tb": {
        "aliases": {
            "addr2line": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-addr2line.exe",
            "ar": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-ar.exe",
            "as": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-as.exe",
            "c++filt": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-c++filt.exe",
            "g++": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-g++.exe",
            "gcc": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-gcc.exe",
            "ld": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-ld.exe",
            "nm": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-nm.exe",
            "objcopy": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-objcopy.exe",
            "objdump": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-objdump.exe",
            "ranlib": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-ranlib.exe",
            "readelf": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-readelf.exe",
            "size": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-size.exe",
            "strip": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-strip.exe"
        },
        "build_type": "release",
        "cc_flags": [
            "-std=gnu17",
            "-fopt-info-all=optimization_report.txt",
            "-mcpu=cortex-m3",
            "-mthumb",
            "-Os",
            "-ffunction-sections",
            "-fdata-sections",
            "-nostdlib",
            "--param",
            "max-inline-insns-single=500"
        ],
        "cc_path": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-gcc.exe",
        "compiler_type": "gcc",
        "cxx_flags": [
            "-std=gnu++17",
            "-fno-threadsafe-statics",
            "-fno-rtti",
            "-fno-exceptions",
            "-fno-use-cxa-atexit",
            "-fopt-info-all=optimization_report.txt",
            "-mcpu=cortex-m3",
            "-mthumb",
            "-Os",
            "-ffunction-sections",
            "-fdata-sections",
            "-nostdlib",
            "--param",
            "max-inline-insns-single=500"
        ],
        "cxx_path": "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-g++.exe",
        "defines": [
            "PLATFORMIO=60116",
            "STM32F1",
            "STM32F103xB",
            "STM32F1xx",
            "ARDUINO=10808",
            "ARDUINO_ARCH_STM32",
            "NDEBUG",
            "ARDUINO_HY_TINYSTM103TB",
            "BOARD_NAME=\"HY_TINYSTM103TB\"",
            "HAL_UART_MODULE_ENABLED",
            "USE_HAL_DRIVER",
            "USE_FULL_LL_DRIVER",
            "VARIANT_H=\"variant_HY_TINYSTM103TB.h\"",
            "BL_LEGACY_LEAF",
            "VECT_TAB_OFFSET=0x2000"
[... 51 lines cut ...]
                "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\arm-none-eabi\\include\\c++\\12.3.1\\arm-none-eabi",
                "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\12.3.1\\include",
                "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\12.3.1\\include-fixed",
                "C:\\Users\\niteris\\.platformio\\packages\\toolchain-gccarmnoneeabi\\arm-none-eabi\\include"
            ]
        },
        "libsource_dirs": [
            "C:\\Users\\niteris\\dev\\fastled\\.build\\hy_tinystm103tb\\lib",
            "C:\\Users\\niteris\\dev\\fastled\\.build\\hy_tinystm103tb\\.pio\\libdeps\\hy_tinystm103tb",
            "C:\\Users\\niteris\\.platformio\\lib",
            "C:\\Users\\niteris\\.platformio\\packages\\framework-arduinoststm32\\libraries\\__cores__\\arduino",
            "C:\\Users\\niteris\\.platformio\\packages\\framework-arduinoststm32\\libraries"
        ],
        "prog_path": "C:\\Users\\niteris\\dev\\fastled\\.build\\hy_tinystm103tb\\.pio\\build\\hy_tinystm103tb\\firmware.elf",
        "svd_path": "C:\\Users\\niteris\\.platformio\\platforms\\ststm32\\misc\\svd\\STM32F103xx.svd",
        "targets": [
            {
                "group": "Platform",
                "name": "upload",
                "title": "Upload"
            }
        ]
    }
}
```

zackees, 2025-03-03:
And the build output:

```
In file included from C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:3:
src/FastLED.h:837:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
  837 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
      |  ^~~~~~~
Compiling .pio\build\hy_tinystm103tb\libca7\src\FastLED.cpp.o
In file included from src/FastLED.h:83:
src/fastspi.h:171:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  171 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged 
output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/FastLED.h:66:
src/fastpin.h: In instantiation of 'class FastPin<1>':
src/fastspi_bitbang.h:27:49:   required from 'class AVRSoftwareSPIOutput<1, 2, 12>'
src/fastspi.h:51:7:   required from 'class SPIOutput<1, 2, 12>'
src/chipsets.h:373:6:   required from 'class APA102Controller<1, 2, RGB, 12, kFiveBitGammaCorrectionMode_Null, 0, 4278190080>'
src/FastLED.h:478:34:   required from 'CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = APA102; unsigned char DATA_PIN = 1; unsigned char CLOCK_PIN = 2; EOrder RGB_ORDER = RGB]'
C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:16:66:   required from here
src/fastpin.h:289:31: error: static assertion failed: Invalid pin specified
  289 |         static_assert(validpin(), "Invalid pin specified");
      |                       ~~~~~~~~^~
src/fastpin.h:289:31: note: 'FastPin<1>::validpin()' evaluates to false
src/fastpin.h: In instantiation of 'class FastPin<2>':
src/fastspi_bitbang.h:28:50:   required from 'class AVRSoftwareSPIOutput<1, 2, 12>'
src/fastspi.h:51:7:   required from 'class SPIOutput<1, 2, 12>'
src/chipsets.h:373:6:   required from 'class APA102Controller<1, 2, RGB, 12, kFiveBitGammaCorrectionMode_Null, 0, 4278190080>'
src/FastLED.h:478:34:   required from 'CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = APA102; unsigned char DATA_PIN = 1; unsigned char CLOCK_PIN = 2; EOrder RGB_ORDER = RGB]'
C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:16:66:   required from here
src/fastpin.h:289:31: error: static assertion failed: Invalid pin specified
src/fastpin.h:289:31: note: 'FastPin<2>::validpin()' evaluates to false
In file included from src\FastLED.cpp:2:
src\FastLED.h:837:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
  837 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
      |  ^~~~~~~
*** [.pio\build\hy_tinystm103tb\src\Apa102.ino.cpp.o] Error 1
```