FastLED/FastLED #1891: MG24 support request (Seeed Xiao MG24 Sense)

punkweazel, 2025-02-17:
I wonder if it would be possible to add support for FastLED to work on the new Seeed Xiao MG24 Sense board?
Product page:  https://www.seeedstudio.com/Seeed-XIAO-MG24-Sense-p-6248.html
Using the Silicone Labs EFR32MG24 Series 2 Multiprotocol Wireless SoC - https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-socs

The last time I checked, the Neopixel library doesn't support it either.  It has an in-house LED driver library that we've been using, but I would obviously prefer to be able to use FastLED.
Silicone Labs ezWS2812 library:   https://github.com/SiliconLabs/arduino/tree/b0492674316ec6505a474155a2bbc3b3e9446417/libraries/ezWS2812

The XIAO MG24 is a pretty sweet board.  I think people would appreciate being able to use FastLED if someone was keen to add support for it, so I'm just putting this out there.  I'm not experienced with library coding, but maybe I could help in ways.

Thanks

zackees, 2025-03-11:
Any way you can provide a `platformio.ini` file?

This is how we do bringup.

punkweazel, 2025-03-11:
Thanks for the replies.

Yeh I understand it's a community contribution vibe.

I haven't used platformIO with the MG24 yet... I was planning to eventually.  Not sure if it's supported there yet either, as I know the Xiao nrf52xxx took a while.

For now we're doing okay with an LED driver library that my friend built mostly from scratch.  I always wanted to use FastLED for the noise function, among others. I can get noise/math functions elsewhere though.

As our project gets more complicated, I'd like to move to platformIO and also provide whatever info I can to contribute to FastLED, even if we don't use it.  I'll check in on this thread every now and then.

zackees, 2025-03-11:
I untangled the noise functions from FastLED.h i believe. Actually i've untangled a lot of other code too like CRGB.

You should just try including the library but not using the driver portion.

punkweazel, 2025-03-12:
Oh cool.  Thanks.  I'll check that out.

zackees, 2025-09-26:
Fixed in https://github.com/FastLED/FastLED/commit/047ac08d2054816d7ffe60200bea8f9c6b2ea418

You can use define `FASTLED_USES_EZWS2812_SPI` before `#include <FastLED.h` to get the optimized SiliconLabs WS2812 driver using spi. Both the GPIO and SPI driver are now part of fastled.

closed 2025-09-26 by zackees