FastLED/FastLED #1900: Doesn't compile if FASTLED_NAMESPACE is defined

Rob190, 2025-03-08:
I need FastLED to be in a namespace due to multiple conflicting names.

If I define FASTLED_NAMESPACE:

`#define FASTLED_NAMESPACE Fastled`

I get a number of compilation errors, because some of the code assumes the names are not in a namespace e,g, in xymap.h:

`    void mapPixels(const CRGB *input, CRGB *output) const;`

`CRGB `would need to be `Fastled::CRGB`

There are a few other names that have the same issue.

zackees, 2025-03-08:
We have one supported namespace and that is `fl`. Are you saying you have another library that uses the same `fl` namespace as our library?

Rob190, 2025-03-08:
OK so maybe I don't understand how your system works. I was expecting it to behave the same as the C++ standard library where you include the header and then use the qualified name or a 'using namespace' clause. It appears you default to no namespace but then add it by defining FASTLED_NAMESPACE. That's fine (although maybe a little unexpected). I just tried `#define FASTLED_NAMESPACE f1` but get the same errors.

I think it would be a good idea to use a longer namespace name than `f1`. I don't have any namespaces of that name but I do actually have a variable named `f1` and I can imagine it isn't that unusual.

zackees, 2025-03-08:
Until recently we had no default namespace.

Old sketches expect there not to be a namespace. New code is namespaced and put into the `src/fl` folder. With certain defines, the entire repo uses fl namespace.

This is done so that devs don't get header & symbol collisions in the global namespace.

closed 2025-03-08 by zackees

Rob190, 2025-03-08:
Fully understand the issue with old apps not using namespaces. However, I need to use a namespace and, in the absence of any documentation, I tried what I said above but it didn't compile.

Is there an example that shows how to enable the f1 namespace? Otherwise I will have to resort to using a locally-modified copy of the library.

zackees, 2025-03-08:
This is the build define:

FASTLED_FORCE_NAMESPACE=1

Our tests use this so that the entire library is wrapped in the fl namespace.

https://github.com/FastLED/FastLED/blob/master/tests/CMakeLists.txt

Rob190, 2025-03-09:
It still doesn't work for me. E.g. compiling the NoisePlusPalette example with` build_flags = -DFASTLED_FORCE_NAMESPACE=1` results in a lot of errrors including:

```
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/rmt_4/idf4_rmt.cpp:16:6: error: 'RmtController' has not been declared
 void RmtController::init(int pin, bool built_in_driver) {
      ^~~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/rmt_4/idf4_rmt.cpp:21:1: error: 'RmtController' does not name a type
 RmtController::RmtController(int DATA_PIN, int T1, int T2, int T3,
 ^~~~Compiling .pio\build\esp32-c3-devkitm-1\lib6d9\FastLED\platforms\wasm\engine_listener.cpp.o
~~~~~~~~~
```

In order to get my app working, I've disabled this flag, moved all the FastLED code into a separate file and put it inside a wrapper layer. This removes the conflicts at the expense of some duplication of code. It's not ideal because the pin numbers are template parameters, so there is no way to allow the app to specify them without including FastLED and I can't do that in the app due to the conflicts. Consequently, the pin numbers need to be in the wrapper and so the wrapper is app-specific.

zackees, 2025-03-10:
Okay this actually is a bug. I'll fix it.

zackees, 2025-03-11:
Okay this is now tested in every cl.

Here is your status badge:

[image]