FastLED/FastLED #1902: variable led number

jan5650, 2025-03-15:
Hello!

How can I make it so that in the setup section in EEPROM I can read how many LEDs I want to control, and the FastLED library only uses that many?

#define NUM_LEDS 40

This would be variable 40. Sometimes 20, sometimes 30. The problem is that I can only read from EEPROM in the setup section, but the CRGB leds[NUM_LEDS]; section is in front of the setup, so I don't know how many LEDs I want to control

is there a solution to this? Is it possible to redefine the CRGB leds[NUM_LEDS]; later somehow?

Thank you!

sutaburosu, 2025-03-15:
Yes. [See here](https://www.reddit.com/r/FastLED/comments/16bum9z/ways_to_reduce_memory_footprint_of_led_array/jzfliwq/). For support using FastLED, that subreddit is usually the best place to ask. This issue tracker is for problems and improvements in FastLED itself.

zackees, 2025-03-19:
your leds can be variable. Replace the array with a pointer and the rest of your code can provably work fine

closed 2025-03-19 by zackees