FastLED/FastLED #1904: FastLed and ESP-IDF

andrewjswan, 2025-03-18:
I did not find any detailed information about the compatibility of the current version of FastLed and ESP-IDF 5+, I tried to build the project, but ESP-IDF does not find a compatible library, if I change:

https://github.com/FastLED/FastLED/blob/4ab0c7282ebb1f3ae75f8ffc6d7240e634638f59/library.json#L44

to `"frameworks": "*",` , it finds it, but errors occur during assembly.

There is no compatibility? Or I don't fully understand how it works?

zackees, 2025-03-19:
Our FastLED build binds to arduino core esp32.

You'll need that in order to build.

If you make profess please share what you did

andrewjswan, 2025-03-19:
Got it, the question is closed.

closed 2025-03-19 by andrewjswan