FastLED/FastLED #1905: Giga R1 still not supported in Arduino IDE installed 3.9.14

k1ngcyk, 2025-03-19:
Checked the source code installed by IDE, and also checked GitHub's release zip for 3.9.14 and the tag `3.9.14`. No `ARDUINO_GIGA` found in `src/led_sysdefs.h`.

sutaburosu, 2025-03-19:
There was [a change to Giga support](https://github.com/FastLED/FastLED/commit/01588a45bbf7956787cd3258074fb71719c57005) after 3.9.14 was tagged and released. Until the next release, you could download [a .zip of master](https://github.com/FastLED/FastLED/archive/refs/heads/master.zip) and extract it to the same location where your IDE installed 3.9.14.

zackees, 2025-03-19:
Exactly what @sutaburosu said. We have one more fix for the Giga R1 that will go in with 3.9.15 of fastled. I'm waiting on some other fixes to go in before I mint a new version. So please just use the master branch zip.

Or you can use branch `stable` if you are concerned about some fresh unstable fix making it into your build.

phenixsims, 2025-04-04:
I tried the master and stable versions and I get an error trying to verify the sketch.

```
C:\Users\[name]\Documents\Arduino\libraries\FastLED\src\noise.cpp:16:63: warning: variable length array 'H' is used [-Wvla]
 #define VARIABLE_LENGTH_ARRAY(TYPE, NAME, SIZE) TYPE NAME[SIZE]
                                                               ^
C:\Users\[name]\Documents\Arduino\libraries\FastLED\src\noise.cpp:871:3: note: in expansion of macro 'VARIABLE_LENGTH_ARRAY'
   VARIABLE_LENGTH_ARRAY(uint8_t, H, height*width);
   ^
exit status 1
Error compiling for board Arduino Giga R1.
```

zackees, 2025-04-07:
Will go out in the next release which is expected to be tomorrow.

closed 2025-04-07 by zackees