FastLED/FastLED #1908: Brightness of 8 or less treated as 0 when driving SK9822 (and likely APA102)

starduststorm, 2025-03-23:
I have noticed that FastLED will draw nothing at all if brightness is set to 8 or less when driving SK9822 (or presumably APA102) pixels. This is true regardless of whether or not FASTLED_USE_GLOBAL_BRIGHTNESS is on. But I'm not on AVR and it now seems that this flag is entirely preempted by FASTLED_HD_COLOR_MIXING which is on by default if not on AVR.

Occurs with FastLED 3.9.14
behavior introduced with e44f1820ef0baab0b4c38a421ae9361d32920d54 (discussion in https://github.com/FastLED/FastLED/issues/1709)
Repro: rp2040 SK9822
Not repro: rp2040 SK9822HD
Repro: teensy3.2 SK9822
Not repro: teensy3.2 WS2812B

getGlobalBrightnessAndScalingFactors is called by showPixelsDefault. this line clips the value: *out_brightness = map(brightness, 0, 255, 0, 31);

This behavior occurs on SK9822, which is how I discovered that there is now SDK9822HD, where is does not repro.

Apologies as I am not very familiar with the new FastLED HD variants of SK9822 and APA102, and the plan for how this is all working for multibyte channels. It seems maybe that it's computing the 5-bit brightness field per pixel instead of globally? If so awesome.

In any case though it does seem incorrect that brightness would get clipped down to zero in the legacy global brightness case, rather than down to 1 until FastLED.setBrightness is actually 0.

Thanks!

```
#include <Arduino.h>
#include <FastLED.h>

CRGBArray<8> leds;

void setup() {
  Serial.begin(57600);
  while (!Serial) delay(10);
  delay(2000);
  FastLED.addLeds<SK9822, SPI_DATA, SPI_CLOCK, BGR, DATA_RATE_MHZ(16)>(leds, leds.size());
  // FastLED.addLeds<WS2812B, 11, GRB>(leds, 8);
}

void loop() {
  Serial.println("loop");
  leds[0] = CRGB::White;
  leds[1] = CRGB::Red;
  leds[2] = CRGB::Green;
  leds[3] = CRGB::Blue;
  leds[4] = CHSV(millis()/10,0xFF,0xFF);

  FastLED.setBrightness(8);
  FastLED.show();
// nothing is shown
}

```

zackees, 2025-04-06:
Yes, the global brightness is now done via the 5-bit brightness factor in the APA102/SK9822 led chipset. This truncates the total range of brightness values like you are seeing here. However, it boosts the color fidelity all the way down to the lowest setting. If you apply global brightness to the WS2812 which doesn't have this, you'll notice that color components get lopped off as the LEDs go down to the lowest brightness settings.

You'll probably also notice that the global brightness stair steps and doesn't respond to all every global brightness setting.

It's a tradeoff between preserving color values at all brightness settings vs letting color components completely shut off.

Here are my ideas for brightness improvements:

1. Shift the brightness factor so that it works all the way from 255->1, such 255 to 240 are all map to 31 (right now it's 255->248 which map to 31). This would be high side truncation. This would probably be better than the low side truncation that you are seeing now.

or 2. Split the brightness factor between the 5-bit brightness and the global brightness RGB scaling component that is sent down to the LED strip, if and only if the brightness factor is <= 8.

I've got higher priority things to tackle so I can't get to this issue right away.

Also, I may recommend that you try using APA102HD or SK9822HD modes. This does advanced color mixing and driver level gamma correction and looks amazing, hence the "HD" mode specifier.

If you improve this color mode for the non HD variants, I will gladly accept a PR.

zackees, 2025-04-08:
Live now in the master branch. Feel free to download the repo as a zip and install it manually in ArduinoIDE

closed 2025-04-08 by zackees