FastLED/FastLED #1914: ESP32C2 CORRUPT HEAP: Bad head at 0x3fca7520. Expected 0xabba1234 got 0x00000003

blueskytomorrow, 2025-04-02:
I am using ESP32C2 with BLE and FastLED. I encounter wired issue causing CORRUPT HEAP: Bad head.

case 1:
if I put the BLE initialized function (wich is just standard arduino BLE functions) in the first half of the setup() function, call FastLED.show() many times, it works well.

e.g 
void setup() {
  Serial.begin(115200);
  pinMode(FAN_PIN, OUTPUT);
  analogWrite(FAN_PIN,0);
  initBLE();

  // 灯初始化
  FastLED.addLeds<WS2812, RGB_PIN,GRB>(leds, NUM_LEDS);
  // 刷新 LED 显示;  // GRB ordering is assumed
  FastLED.setBrightness(50);  // 设置亮度
  fill_solid(leds, NUM_LEDS,CRGB::Red);
  FastLED.show();
  delay(3000);
  fill_solid(leds, NUM_LEDS,CRGB::Green);
  FastLED.show();
}

case 2:
if I put some hardware related logic, like analogWrite or even analogRead, the chip errors out when the second FastLED.show() is called
CORRUPT HEAP: Bad head at 0x3fca7520. Expected 0xabba1234 got 0x00000003

e.g
void setup() {
  Serial.begin(115200);
  pinMode(FAN_PIN, OUTPUT);

  // 灯初始化
  FastLED.addLeds<WS2812, RGB_PIN,GRB>(leds, NUM_LEDS);
  // 刷新 LED 显示;  // GRB ordering is assumed
  FastLED.setBrightness(50);  // 设置亮度
  fill_solid(leds, NUM_LEDS,CRGB::Red);
  FastLED.show();
  delay(3000);
  analogWrite(FAN_PIN,0);
  fill_solid(leds, NUM_LEDS,CRGB::Green);
  FastLED.show();
initBLE();
}

case 3:
for BLE callback, if I put FastLED.show() in the call back and the received string strValue is above certain length e.g 12, it will trigger heap error as well for the second call to FastLED.show()

e.g.
void onWrite(BLECharacteristic* pCharacteristic) {
    String strValue = pCharacteristic->getValue(); 
    FastLED.show()
}

case 4:
if I put FastLED.show() in the loop() function, it there is any pin related read/write, the FastLED.show() will error out as well.

I have used 

#define FASTLED_ESP32_USE_CLOCKLESS_SPI
#include <FastLED.h>

anyone has the same issue?guess it is related to the DMA and SPI functions

zackees, 2025-04-06:
Can you build with platformio and add this to your platformio.ini file? This will enable stack traces + decoding and we can get a better idea on whats going on

```
monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

build_type = debug

build_flags =
    -DDEBUG
    -g
    -Og
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE
```

blueskytomorrow, 2025-04-27:
> Can you build with platformio and add this to your platformio.ini file? This will enable stack traces + decoding and we can get a better idea on whats going on
>
> ```
> monitor_filters =
> 	default [...]

I am using arduino IDE,but I will have a try on platformio. will let you know once it is done.

zackees, 2025-04-28:
Yeah ArduinoIDE doesn't give you stack traces and has poor defaults for debugging in general.

closed 2025-08-24 by zackees