FastLED/FastLED #1921: FASTLED_USES_ESP32S3_I2S with WS2813

Pertempto, 2025-04-17:
Does the `FASTLED_USES_ESP32S3_I2S` flag only work for WS2812? Is there any way to use it with the WS2813?

I tried following the example from [here](https://github.com/FastLED/FastLED/blob/master/examples/Esp32S3I2SDemo/Esp32S3I2SDemo.ino#L111)

```cpp
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA0, GRB>(
        leds + (0 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA1, GRB>(
        leds + (1 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA2, GRB>(
        leds + (2 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA3, GRB>(
        leds + (3 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA4, GRB>(
        leds + (4 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA5, GRB>(
        leds + (5 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA6, GRB>(
        leds + (6 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA7, GRB>(
        leds + (7 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA8, GRB>(
        leds + (8 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA9, GRB>(
        leds + (9 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA10, GRB>(
        leds + (10 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.addLeds<WS2813, EXAMPLE_PIN_NUM_DATA11, GRB>(
        leds + (11 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP);
    FastLED.setBrightness(32);
```

...but that only resulted in crashes with an error when we call `FastLED.show()`

```
E (735) rmt: rmt_tx_register_to_group(132): no free tx channels
E (738) rmt: rmt_new_tx_channel(239): register channel failed
E (744) led_strip_rmt: led_strip_new_rmt_device(192): create RMT TX channel failed
```

When we try using the driver directly with ` fl::InternalI2SDriver::create()` and `driver->initled`, the LED output seems "squished". Drawing a 12x12 array of white pixels results in a rectangle that is only 9 pixels wide. This made us think that we need to change something with the timings.

Any help with this would be greatly appreciated!

zackees, 2025-05-06:
Hi, you can support different WS28XX chipsets via an overclocking factor. I just added this for you and it's only on the main branch so you'll need to manually install it.

The overclocking factor works because it compresses or expands the T0H value, the rest of the LED chipset timing values are actually very flexible.

The #define is `FASTLED_LED_OVERCLOCK`. The default value is 1.0, while 1.2 represents a 20% overclock. Try .9 or 1.1 in your project.

However, this define is a BUILD LEVEL DEFINE and not an include define. If you are using Arduino IDE then you'll need to edit the source code and set the value to what you want it to be, or else use platformio. I recommend using platformio, which has a section for build_flags, which you can set to something like `build_flags=-DFASTLED_LED_OVERCLOCK=1.1`

If you aren't using platformio, then check out our handy guide: https://github.com/FastLED/PlatformIO-Starter

Switching to platformio is way easier than most users think it will be. And your project, if you follow our template, will be compatible with the ArduinoIDE (except for the build define).

Pertempto, 2025-05-06:
Thanks @zackees! We are already using PlatformIO for our project. In the past few weeks we back to using https://github.com/hpwit/I2SClockLessLedDriveresp32s3 directly instead of FastLED. The information about tinkering with `CLOCKLESS_PIXEL_CLOCK_HZ` and `T0H` is very helpful, we'll try that.

closed 2025-05-06 by Pertempto