FastLED/FastLED #1922: I2SClocklessLedDriveresp32S3.setBrightness only uses _gammag?

Hypfer, 2025-04-18:
While trying to understand how to get my matrix darker, I just stumbled upon this code
https://github.com/FastLED/FastLED/blob/e146a17b65f51a97a8825c6ab3b99d7f7defed68/src/third_party/yves/I2SClockLessLedDriveresp32s3/src/I2SClockLessLedDriveresp32s3.h#L283-L298

and I'm wondering if that should be `_gammag` everywhere or if that is just a copy-paste error.

I'm _also_ wondering what will break further downstream _if_ this is indeed wrong and would be changed to reference `_gammab`, `_gammar` and `_gammaw` instead.

Hypfer, 2025-05-29:
Unfortunately, I do not have a setup that could be used to validate and test this.
I've just been playing around with Awtrix3 and not even building it on my own machine but in GitHub actions.
Hence this became just an issue instead of a PR.

My assumption was that someone from the team probably has some test thing lying around, which does all sorts of edge cases and which they've seen so often, they'd easily be able to spot minor differences to how it was before.

zackees, 2025-05-30:
Confirmed, Yves said it already correct on his repo so I'll fix this to match.