FastLED/FastLED #1923: ESP32-S3 - RMT Issues beyond Release 3.9.0

louisnorwood, 2025-04-19:
Hello,

I have been locked into using version 3.9.0 for a while now, as I am using an ESP32-S3, and doing a bunch of other tasks on the processor.

On every version beyond 3.9.0 (since RMT5 updates) if I iniitialize more than one LED strip then I get this error below, and get constant reboots during setup. 

The latest released version is currently 3.9.16, and thi works fine as long as I only intialize one pin of LEDs.
-----------------------------------
ERROR MESSAGE WITH BOOTLOOP:
E (112) rmt: rmt_tx_register_to_group(132): no free tx channels
E (112) rmt: rmt_new_tx_channel(239): register channel failed
E (113) led_strip_rmt: led_strip_new_rmt_device(192): create RMT TX channel failed
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42018c91
file: "c:\Users\louis\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\rmt_5\strip_rmt.cpp" line 66
func: led_strip_t* {anonymous}::configure_led_with_timings(int, uint32_t, bool, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, bool, uint8_t)
expression: led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip)

abort() was called at PC 0x4037e51b on core 1

-------------------------------------------

I have tried both the following lines before including FastLED, but this does not seem to affect the issue:

#define FASTLED_RMT_MAX_CHANNELS 1
#define FASTLED_RMT_BUILTIN_DRIVER 1

I would like to create 8 pins in FastLED. This works on the old version wtih all eight pins, however now I'm using ArduinoJson eleswhere in the project, I get weird clashes with the libraries due to the included /third_party/arduino-json code.

So I'm hoping I can find a solution to getting the latest version working, maybe by disabling RMT or some other work-around.

Any suggestions of what I can try?
Thanks

4wheeljive, 2025-05-14:
I'm on the lookout too for any fixes/workarounds on this. 

@louisnorwood FWIW, I am also using an ESP32-S3 but I'm able to drive up to 4 LED strips on separate pins before getting reboot/error issues similar to yours with anything after 3.9.0 (i.e., since RMT5). I'm happy to share any info about my setup/sketch that might help you get from 1 to 4. 

Here are some other posts I've seen that involve this issue:
- https://github.com/espressif/idf-extra-components/issues/473
- https://github.com/FastLED/FastLED/issues/1768
- https://github.com/espressif/idf-extra-components/issues/491

(NOTE: I'm pretty much a noob on all of this: FastLED/Arduino/GitHub/coding/electronics/etc. So please forgive and gently correct me regarding any etiquette/protocol transgressions!)

zackees, 2025-05-29:
Bro, you have an S3, why aren't you using the I2S driver? It's WAY more stable than RMT.

[image]

You don't even have to use all the pins, you can use as few as you want.

louisnorwood, 2025-05-30:
@zackees. Mostly becuase I didn't want to move into PlatformIO compilation, had loads of issues with library stuff last time I went down that rabbit hole. 
The last time I tried that I2S functionallity from Ardiuno IDE, even the examples just bootlooped for me. probably to do with PSRAM stuff I didn't understand. Had been trying to get the working from Yves repository for a while before you integrated it.

I got things stable with Async and other libs up to ESP core3.0.7. And platformIo uses a much older core, and the previous IDF version meant I needed to go down a bunch of other rabbit holes.

Does it work in Arduino IDE? or have you got any advice for setting up a PlatformIO that works. I saw you recommend cloning your lib for someone else, but tried that when that version went live, and didn't work for me. 
I have seen your last couple of releases seemed to have bug fixed I2S, so maybe just paying the price for attempting to be an early adopter.

zackees, 2025-05-31:
We created a template project for platformIO that will compile with both ArduinoIDE and PlatformIO without having to change anything in your source code.

Just follow the step by step directions here:

https://github.com/FastLED/PlatformIO-Starter

zackees, 2025-05-31:
Also, our version of the Yves drive is much easier to use. You don't have to have a rectangular buffer when you use our version of the driver. You just have a bunch of strips of any length defined, and the buffer is made rectangular behind the scenes. So the API for I2S now matches that for RMT.

louisnorwood, 2025-06-01:
@zackees I am impressed with your platformIO-Starter project, it did make it very easy to get this built. Also including the necessary declaration settings for each type of S3 chip is very helpful - Only suggestion is to add this line to platformio.ini, so the serial monitor works.
`monitor_speed = 115200`

This tool is much faster to complie (and re-compile cahnges) which is my main frustration with using ESP32s and ArduinoIDE
----
Still having issues with the I2S driver sadly, initially I was using some pins that clashed as I am using the OPI PSRAM (18Mb), which uses pins 35,36&37, that was causing reboots within the loop, but once I found this, I now have my 5 remaining LED outputs active (pins38-42), but am not in control of the LEDs as expected, I have used FastLED for many years, and am a fairly experienced programmer and debugger,  so hoping this isn't just me being very stupid.
----
I was getting strange white flickering when using the I2S example fill_rainbow() function. So have simplified it to just attempt to set all LEDs off and the first to Blue.
`  void loop() {
    //fill_rainbow(leds);
    for( int i=0; i<NUM_LEDS; i++){
      leds[i] = CRGB(0,0,0);
    }
    leds[0] = CRGB(0,0,127);
    FastLED.show();
}
`

This still gives me white outputs on all 5 pins. Stripping it back to a single pin, I still get white on that one (pin42), but obviosuly not on the disabled ones

My suspicion was that maybe the pins I am using for I2S are not compatible, although I have used them with other libraries apparently using I2S function, and they work successfully. So I hacked a wire into pin 17, but get the same white output from it.

If you have any ideas of things I can test, that would be much appreciated.

Relevant code below for your review, thank you for your time.
platformio.ini
`;ESP32-S3-WROOM-(1/1U)-N16R8 - Flash: 16MB QD, PSRAM: 8MB OT
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
board_build.extra_flags = 
	-DBOARD_HAS_PSRAM
	-DCORE_DEBUG_LEVEL=4

monitor_speed = 115200

lib_deps =
	FastLED `

Main.cpp
`#include <esp_psram.h>
#define FASTLED_USES_ESP32S3_I2S
#include "FastLED.h"
#include "fl/assert.h"

#define NUMSTRIPS 5
#define NUM_LEDS_PER_STRIP 256
#define NUM_LEDS (NUM_LEDS_PER_STRIP * NUMSTRIPS)

#define LEDPin1 17  
#define LEDPin2 41 
#define LEDPin3 40  
#define LEDPin4 39  
#define LEDPin5 38  
#define LEDPin6 37  //Problem pin - OPI RAM Conflict
#define LEDPin7 36  //Problem pin - OPI RAM Conflict
#define LEDPin8 35  //Problem pin - OPI RAM Conflict

int PINS[] = {LEDPin1,LEDPin2,LEDPin3,LEDPin4,LEDPin5};
CRGB leds[NUM_LEDS];

void setup_i2s_using_fastled_api() {
    FastLED.addLeds<WS2812, LEDPin1, GRB>( leds + (0 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP );
    FastLED.addLeds<WS2812, LEDPin2, GRB>( leds + (1 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP );
    FastLED.addLeds<WS2812, LEDPin3, GRB>( leds + (2 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP );
    FastLED.addLeds<WS2812, LEDPin4, GRB>( leds + (3 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP );
    FastLED.addLeds<WS2812, LEDPin5, GRB>( leds + (4 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP );
    FastLED.setBrightness(32);
}

void setup() {
    psramInit(); 
    Serial.begin(115200);

    log_d("Total heap: %d", ESP.getHeapSize());
    log_d("Free heap: %d", ESP.getFreeHeap());
    log_d("Total PSRAM: %d", ESP.getPsramSize());  // If this prints out 0, then PSRAM is not enabled.
    log_d("Free PSRAM: %d", ESP.getFreePsram());
    log_d("waiting 6 second before startup");
    delay(6000);  

    setup_i2s_using_fastled_api();
    log_d("Setup Completed");
}

void loop() {
    for( int i=0; i<NUM_LEDS; i++){
      leds[i] = CRGB(0,0,0);
    }
    leds[0] = CRGB(0,0,127);
    
    FastLED.show();
}`

zackees, 2025-06-02:
Thank you. I will forward to Yves to see what he has to say.

zackees, 2025-06-10:
From the readme:

Note some users find that newer versions of the ESP32 Arduino core (3.10) don't work very well, but older versions do, see issue 1903

You need to downgrade your arduino core. This can be done in both Arduino and PlatformIO.

Our repo PlatformIO-Starter is pinned to a correct version I believe.

closed 2025-06-24 by zackees