FastLED/FastLED #1925: Support for RP2350B (GPIO 30-47)

sfe-SparkFro, 2025-05-07:
The RP2350B has 48 GPIO pins, compared to the 30 GPIO pins of the RP2350B and RP2040. Because only GPIO 0-29 are hard coded, it's not possible to control LEDs on GPIO 30-47. For example, using GPIO 37 results in the following error:

```fastpin.h:289:31: note: 'FastPin<37>::validpin()' evaluates to false```

The root problem appears to be here:

https://github.com/FastLED/FastLED/blob/ef2cb496082d019dbe8774ce2ae5124ed9265117/src/platforms/arm/rp2040/fastpin_arm_rp2040.h#L48-L56

Need to detect if the RP2350B is in use, and if so, add the additional GPIO pins. Also need to do some work to use `uint64_t` instead of `uint32_t` (`uint`), since there is some bit shifting/masking per pin.

zackees, 2025-05-08:
If you help me I'll help you out.

See those build status icons on the FastLED readme page? If you click through it you'll see a build step the dumps the json of the build state, including the build defines, like this:

[image]

Our build system uses platformio. If you can get the simplest of platformio.ini and get it to compile a sketch, then I will put a build badge for this model. And then we will get the defines.

These defines will allow you to fingerprint the build environment and special case your code.

sfe-SparkFro, 2025-05-09:
I rarely use PlatformIO, but I gave it a whirl with the VS Code plugin.

platformio.ini:

```
[env:sparkfun_xrprp2350]
platform = raspberrypi
board = sparkfun_xrprp2350
framework = arduino
lib_deps =
  fastled/FastLED @ ^3.9.16
```

I copied the `blink.ino` example from FastLED into `main.cpp`. Compiling as-is works fine. Changing `DATA_PIN` to `37` results in the same error `'FastPin<37>::validpin()' evaluates to false`

I see the `build_info.json` when clicking through the badge on the FastLED readme as you say, though I'm unable to find that file locally on my end.

fivemhostingserver, 2025-05-09:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemtebex, 2025-05-09:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemtebex, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.

fivemeyestore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

scriptforfivem, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

zackees, 2025-05-10:
> I rarely use PlatformIO, but I gave it a whirl with the VS Code plugin.
>
> platformio.ini:
>
> ```
> [env:sparkfun_xrprp2350]
> platform = raspberrypi
> board = sparkfun_xrprp2350
> framework = arduino
> lib_deps [...]

Are you doing anything special? It doesn't build on windows. What's your operation system?

[image]

zackees, 2025-05-10:
You indicated that this was rp2350B, but I don't see any details about that. I think you just have a 2350.

```json
"defines": [
            "PLATFORMIO=60118",
            "ARDUINO_RASPBERRY_PI_PICO_2",
            "ARDUINO_ARCH_RP2350",
            "USBD_MAX_POWER_MA=250",
            "ARDUINO=10810",
            "ARDUINO_ARCH_RP2040",
            "F_CPU=133000000L",
            "BOARD_NAME=\"rpipico2\"",
            "CFG_TUSB_MCU=OPT_MCU_RP2040",
            "CFG_TUSB_OS=OPT_OS_PICO",
            "LIB_BOOT_STAGE2_HEADERS=1",
            "LIB_PICO_ATOMIC=1",
            "LIB_PICO_BIT_OPS=1",
            "LIB_PICO_BIT_OPS_PICO=1",
            "LIB_PICO_CLIB_INTERFACE=1",
            "LIB_PICO_CRT0=1",
            "LIB_PICO_CXX_OPTIONS=1",
            "LIB_PICO_DIVIDER=1",
            "LIB_PICO_DIVIDER_COMPILER=1",
            "LIB_PICO_DOUBLE=1",
            "LIB_PICO_FIX_RP2040_USB_DEVICE_ENUMERATION=1",
            "LIB_PICO_FLOAT=1",
            "LIB_PICO_INT64_OPS=1",
            "LIB_PICO_INT64_OPS_COMPILER=1",
            "LIB_PICO_MEM_OPS=1",
            "LIB_PICO_MEM_OPS_COMPILER=1",
            "LIB_PICO_NEWLIB_INTERFACE=1",
            "LIB_PICO_PLATFORM=1",
            "LIB_PICO_PLATFORM_COMPILER=1",
            "LIB_PICO_PLATFORM_PANIC=1",
            "LIB_PICO_PLATFORM_SECTIONS=1",
            "LIB_PICO_RUNTIME=1",
            "LIB_PICO_RUNTIME_INIT=1",
            "LIB_PICO_STANDARD_BINARY_INFO=1",
            "LIB_PICO_STANDARD_LINK=1",
            "LIB_PICO_SYNC=1",
            "LIB_PICO_SYNC_CRITICAL_SECTION=1",
            "LIB_PICO_SYNC_MUTEX=1",
            "LIB_PICO_SYNC_SEM=1",
            "LIB_PICO_TIME=1",
            "LIB_PICO_TIME_ADAPTER=1",
            "LIB_PICO_UNIQUE_ID=1",
            "LIB_PICO_UTIL=1",
            "LIB_TINYUSB_BOARD=1",
            "LIB_TINYUSB_DEVICE=1",
            "PICO_32BIT=1",
            "PICO_BOARD=\"pico2\"",
            "PICO_BUILD=1",
            "PICO_COPY_TO_RAM=0",
            "PICO_CXX_ENABLE_EXCEPTIONS=0",
            "PICO_NO_FLASH=0",
            "PICO_NO_HARDWARE=0",
            "PICO_ON_DEVICE=1",
            "PICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1",
            "PICO_RP2040_USB_DEVICE_UFRAME_FIX=1",
            "PICO_RP2350=1",
            "PICO_USE_BLOCKED_RAM=0",
            "TARGET_RP2350",
            "PICO_RP2350=1",
            "PICO_FLASH_SIZE_BYTES=209[71](https://github.com/FastLED/FastLED/actions/runs/14940885642/job/41977702744#step:11:72)52",
            "LIB_PICO_DOUBLE_PICO=1",
            "LIB_PICO_FLOAT_PICO=1",
            "LIB_PICO_FLOAT_PICO_VFP=1",
            "FILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC",
            "USE_UTF8_LONG_NAMES=1",
            "DISABLE_FS_H_WARNING=1",
            "CFG_TUSB_MCU=OPT_MCU_RP2040",
            "USB_VID=0x2E8A",
            "USB_PID=0x000B",
            "USBD_VID=0x2E8A",
            "USBD_PID=0x000B",
            "USB_MANUFACTURER=\"Raspberry Pi\"",
            "USB_PRODUCT=\"Pico 2\"",
            "PICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1",
            "CYW43_LWIP=1",
            "CYW43_PIO_CLOCK_DIV_DYNAMIC=1",
            "LWIP_IPV4=1",
            "LWIP_IGMP=1",
            "LWIP_CHECKSUM_CTRL_PER_NETIF=1",
            "LWIP_IPV6=0",
            "ARDUINO_VARIANT=\"rpipico2\""
        ],
```

zackees, 2025-05-10:
> The RP2350B has 48 GPIO pins, compared to the 30 GPIO pins of the RP2350B and RP2040. Because only GPIO 0-29 are hard coded, it's not possible to control LEDs on GPIO 30-47. For example, using GPIO 37 results in the following error:
> 
> `fastpin.h:289:31: note: 'FastPin<37>::validpin()' evaluates to false`
> 
> The root problem appears to be here:
> 
> [FastLED/src/platforms/arm/rp2040/fastpin_arm_rp2040.h](https://github.com/FastLED/FastLED/blob/ef2cb496082d019dbe8774ce2ae5124ed9265117/src/platforms/arm/rp2040/fastpin_arm_rp2040.h#L48-L56)
> 
> Lines 48 to 56 in [ef2cb49](/FastLED/FastLED/commit/ef2cb496082d019dbe8774ce2ae5124ed9265117)
> 
>  #define MAX_PIN 29 
>  _FL_DEFPIN(0); _FL_DEFPIN(1); _FL_DEFPIN(2); _FL_DEFPIN(3); 
>  _FL_DEFPIN(4); _FL_DEFPIN(5); _FL_DEFPIN(6); _FL_DEFPIN(7); 
>  _FL_DEFPIN(8); _FL_DEFPIN(9); _FL_DEFPIN(10); _FL_DEFPIN(11); 
>  _FL_DEFPIN(12); _FL_DEFPIN(13); _FL_DEFPIN(14); _FL_DEFPIN(15); 
>  _FL_DEFPIN(16); _FL_DEFPIN(17); _FL_DEFPIN(18); _FL_DEFPIN(19); 
>  _FL_DEFPIN(20); _FL_DEFPIN(21); _FL_DEFPIN(22); _FL_DEFPIN(23); 
>  _FL_DEFPIN(24); _FL_DEFPIN(25); _FL_DEFPIN(26); _FL_DEFPIN(27); 
>  _FL_DEFPIN(28); _FL_DEFPIN(29); 
> Need to detect if the RP2350B is in use, and if so, add the additional GPIO pins. Also need to do some work to use `uint64_t` instead of `uint32_t` (`uint`), since there is some bit shifting/masking per pin.

Anyway, i think you can just use `_FL_DEFPIN(37)`

With platformio you can just right click on of the fastled symbols and click 'Go to definition' and get teleported into the code base. Try adding _FL_DEFPIN(37) and see if it works.

zackees, 2025-05-10:
I went ahead and made a branch with this fix, replace your platformio.ini with this and let me know if it works for you now.

```
[env:sparkfun_xrprp2350]
platform = raspberrypi
board = sparkfun_xrprp2350
framework = arduino

lib_deps =
    https://github.com/FastLED/FastLED.git#sparkfun_2350B_fix ; Pull FastLED from the specific branch
```

zackees, 2025-05-10:
One last thing, you can dump the json like i did by running the following.

First make sure that platformio is installed. You can either open it up through one of the buttons in VSCode (it will drop you into the pio cl) or you can install platformio using `pip install platformio`

Then with that in your environment you can run the following:

```
pio project metadata --json-output
```

sfe-SparkFro, 2025-05-12:
> Are you doing anything special? It doesn't build on windows.

Sorry, I think I had a forked version of the `raspberrypi` platform from an old dead project without realizing it. Haven't touched my PlatformIO install in a while...

Looking into it, the [official `raspberrypi` platform](https://github.com/platformio/platform-raspberrypi) is horribly outdated; it does not support the RP2350 and only includes 2 boards. Reading through some threads, sounds like there was some fallout between PlatformIO and Raspberry Pi (and Espressif), and no external contributions are being accepted into the official platform port. That's disheartening...

However there is [this fork](https://github.com/maxgerhardt/platform-raspberrypi) that actually has the RP2350 and all the latest boards, including the XRP. Here is a corrected platformio.ini that should compile for you:

```
[env:sparkfun_xrp_controller]
platform = https://github.com/maxgerhardt/platform-raspberrypi
board = sparkfun_xrp_controller
framework = arduino
lib_deps = fastled/FastLED @ ^3.9.16
```

> What's your operation system?

Windows

> You indicated that this was rp2350B, but I don't see any details about that. I think you just have a 2350.

Nope, definitely an RP2350B, I designed this board 😉 Just had the wrong platform selected above.

> Anyway, i think you can just use `_FL_DEFPIN(37)`

> With platformio you can just right click on of the fastled symbols and click 'Go to definition' and get teleported into the code base. Try adding _FL_DEFPIN(37) and see if it works.

> I went ahead and made a branch with this fix, replace your platformio.ini with this and let me know if it works for you now.

Nope, doesn't work because the current datatype is `uint32_t`, so this approach only works up to `_FL_DEFPIN(31)`.

```
...
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:46:82: error: right operand of shift expression '(1 << 47)' is greater than or equal to the precision 32 of the left operand [-fpermissive]
   46 | #define _FL_DEFPIN(PIN) template<> class FastPin<PIN> : public _RP2040PIN<PIN, 1 << PIN> {};
      |                                                                                  ^
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:62:17: note: in expansion of macro '_FL_DEFPIN'
   62 | _FL_DEFPIN(46); _FL_DEFPIN(47);
      |                 ^~~~~~~~~~
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:46:82: note: in template argument for type 'long unsigned int'
   46 | #define _FL_DEFPIN(PIN) template<> class FastPin<PIN> : public _RP2040PIN<PIN, 1 << PIN> {};
      |                                                                                  ^
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:62:17: note: in expansion of macro '_FL_DEFPIN'
   62 | _FL_DEFPIN(46); _FL_DEFPIN(47);
```

As I said in my original post:

> need to do some work to use `uint64_t` instead of `uint32_t` (`uint`), since there is some bit shifting/masking per pin.

zackees, 2025-05-21:
What are you doing to get this to compile. What am I doing wrong?

[image]

zackees, 2025-05-21:
Thank you sir for the `platformio.ini` file!

Here is the build badge for the RP2350B [image]

Here is the defines section from the builders `build.json`

```json
"defines": [
            "PLATFORMIO=60118",
            "ARDUINO_SPARKFUN_XRP_CONTROLLER",
            "ARDUINO_ARCH_RP2040",
            "USBD_MAX_POWER_MA=250",
            "PICO_CYW43_SUPPORTED=1",
            "CYW43_PIN_WL_DYNAMIC=1",
            "ARDUINO=10810",
            "ARDUINO_ARCH_RP2040",
            "F_CPU=150000000L",
            "BOARD_NAME=\"sparkfun_xrp_controller\"",
            "CFG_TUSB_MCU=OPT_MCU_RP2040",
            "CFG_TUSB_OS=OPT_OS_PICO",
            "LIB_BOOT_STAGE2_HEADERS=1",
            "LIB_PICO_ATOMIC=1",
            "LIB_PICO_BIT_OPS=1",
            "LIB_PICO_BIT_OPS_PICO=1",
            "LIB_PICO_CLIB_INTERFACE=1",
            "LIB_PICO_CRT0=1",
            "LIB_PICO_CXX_OPTIONS=1",
            "LIB_PICO_DIVIDER=1",
            "LIB_PICO_DIVIDER_COMPILER=1",
            "LIB_PICO_DOUBLE=1",
            "LIB_PICO_FIX_RP2040_USB_DEVICE_ENUMERATION=1",
            "LIB_PICO_FLOAT=1",
            "LIB_PICO_INT64_OPS=1",
            "LIB_PICO_INT64_OPS_COMPILER=1",
            "LIB_PICO_MEM_OPS=1",
            "LIB_PICO_MEM_OPS_COMPILER=1",
            "LIB_PICO_NEWLIB_INTERFACE=1",
            "LIB_PICO_PLATFORM=1",
            "LIB_PICO_PLATFORM_COMPILER=1",
            "LIB_PICO_PLATFORM_PANIC=1",
            "LIB_PICO_PLATFORM_SECTIONS=1",
            "LIB_PICO_RUNTIME=1",
            "LIB_PICO_RUNTIME_INIT=1",
            "LIB_PICO_STANDARD_BINARY_INFO=1",
            "LIB_PICO_STANDARD_LINK=1",
            "LIB_PICO_SYNC=1",
            "LIB_PICO_SYNC_CRITICAL_SECTION=1",
            "LIB_PICO_SYNC_MUTEX=1",
            "LIB_PICO_SYNC_SEM=1",
            "LIB_PICO_TIME=1",
            "LIB_PICO_TIME_ADAPTER=1",
            "LIB_PICO_UNIQUE_ID=1",
            "LIB_PICO_UTIL=1",
            "LIB_TINYUSB_BOARD=1",
            "LIB_TINYUSB_DEVICE=1",
            "PICO_32BIT=1",
            "PICO_BOARD=\"pico2\"",
            "PICO_BUILD=1",
            "PICO_COPY_TO_RAM=0",
            "PICO_CXX_ENABLE_EXCEPTIONS=0",
            "PICO_NO_FLASH=0",
            "PICO_NO_HARDWARE=0",
            "PICO_ON_DEVICE=1",
            "PICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1",
            "PICO_RP2040_USB_DEVICE_UFRAME_FIX=1",
            "PICO_RP2350=1",
            "PICO_USE_BLOCKED_RAM=0",
            "TARGET_RP2350",
            "PICO_RP2350=1",
            "PICO_FLASH_SIZE_BYTES=1677[72](https://github.com/FastLED/FastLED/actions/runs/15152611788/job/42601343340#step:11:73)16",
            "LIB_PICO_DOUBLE_PICO=1",
            "LIB_PICO_FLOAT_PICO=1",
            "LIB_PICO_FLOAT_PICO_VFP=1",
            "FILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC",
            "USE_UTF8_LONG_NAMES=1",
            "DISABLE_FS_H_WARNING=1",
            "CFG_TUSB_MCU=OPT_MCU_RP2040",
            "USB_VID=0x1B4F",
            "USB_PID=0x0046",
            "USBD_VID=0x1B4F",
            "USBD_PID=0x0046",
            "USB_MANUFACTURER=\"SparkFun\"",
            "USB_PRODUCT=\"XRP Controller\"",
            "PICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1",
            "CYW43_LWIP=1",
            "CYW43_PIO_CLOCK_DIV_DYNAMIC=1",
            "LWIP_IPV4=1",
            "LWIP_IGMP=1",
            "LWIP_CHECKSUM_CTRL_PER_NETIF=1",
            "LWIP_IPV6=0",
            "ARDUINO_VARIANT=\"sparkfun_xrp_controller\""
        ],
```

zackees, 2025-05-21:
Here is a platformio with that should work for your setup and has the branch with your boards test fix in it. 

```
[env:sparkfun_xrprp2350]
platform = raspberrypi
board = sparkfun_xrprp2350
framework = arduino
lib_deps =
  fastled/FastLED @ https://github.com/FastLED/FastLED.git#sparkfun_2350B_fix ;
```

Let me know how it goes!

sfe-SparkFro, 2025-05-22:
Thanks! Although, this still does not work for the same reason I said before. Same error message when building:

```
...
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:46:82: error: right operand of shift expression '(1 << 47)' is greater than or equal to the precision 32 of the left operand [-fpermissive]
   46 | #define _FL_DEFPIN(PIN) template<> class FastPin<PIN> : public _RP2040PIN<PIN, 1 << PIN> {};
      |                                                                                  ^
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:62:17: note: in expansion of macro '_FL_DEFPIN'
   62 | _FL_DEFPIN(46); _FL_DEFPIN(47);
      |                 ^~~~~~~~~~
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:46:82: note: in template argument for type 'long unsigned int'
   46 | #define _FL_DEFPIN(PIN) template<> class FastPin<PIN> : public _RP2040PIN<PIN, 1 << PIN> {};
      |                                                                                  ^
.pio\libdeps\sparkfun_xrp_controller\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:62:17: note: in expansion of macro '_FL_DEFPIN'
   62 | _FL_DEFPIN(46); _FL_DEFPIN(47);
```

zackees, 2025-05-23:
Ah, So I have to type def the integer.

zackees, 2025-05-29:
I've been thinking about this for a few days.

I want to ask you a question, does the RP2350B have specific SPI pins it needs to you to use, or is more like a gpio matrix where hardware peripherals can be mapped to any pin you want?

Because if that's the case, maybe it's worth it to consider just removing the FL_DEFPIN macro. This concept came into play because pins were hard wired to peripherals during the AVR days. But that doesn't seem to be the case anymore.

I don't know the answer to this question, do you?

sfe-SparkFro, 2025-05-29:
> I want to ask you a question, does the RP2350B have specific SPI pins it needs to you to use, or is more like a gpio matrix where hardware peripherals can be mapped to any pin you want?

FastLED on the RP2 microcontrollers uses the PIO, not SPI. PIO can be used on any GPIO pins, though there are some things to look out for (eg. the base GPIO pin for each PIO block).

Sorry, but I don't have more time to help with this. I'm just reporting the bug. I might suggest looking other WS2812 drivers for RP2, they have a more "correct" implementation for pin selection and such.

* Raspberry Pi Pico examples: https://github.com/raspberrypi/pico-examples/tree/master/pio/ws2812
* Adafruit NeoPixel: https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_Neopixel_RP2.cpp

If you need a board to test with, [here are a number of boards](https://github.com/search?q=repo%3Aearlephilhower%2Farduino-pico+rp2350b+path%3Avariants&type=code) on the market using the RP2350B that are supported by [Earle Phillhower's Arduino Pico port](https://github.com/earlephilhower/arduino-pico).

zackees, 2025-06-24:
Implemented.