FastLED/FastLED #1930: RAM usage and binary size have exploded on AVR in 3.9.17

sutaburosu, 2025-05-25:
With this release, I run out of RAM when trying to drive a single pixel on a Nano. Just #include'ing FastLED without using it burns over 1KiB of RAM.

```
/*     RAM used (bytes) to show() 1 LED vs FastLED version

|                  | (1) #include | (2) CRGB,addLeds<> | (3) CHSV, show() | Global vars (1/2/3)| Flash used (1/2/3)|
|------------------|--------------|--------------------|------------------|--------------------|-------------------|
| (0) No FastLED   | 206          |                    |                  | 192                | 1908              |
| FastLED 3.9.16   | 223 (+17)    | 306 (+83)          | 334 (+28)        | 209/292/320        | 1946/4314/5234    |
| FastLED 3.9.17   | 1338 (+1134) | ??? crash          | ??? crash        | 869/952/980        | 6424/8792/9712    |

```

[Test sketch and full results](https://wokwi.com/projects/412424157603942401).

nomakewan, 2025-05-25:
Have confirmed this on our Mega 2560 project as well. Also compiling with Arduino IDE 1.8.19:

FastLED 3.9.16
Sketch uses 92368 bytes (36%) of program storage space. Maximum is 253952 bytes.
Global variables use 6041 bytes (73%) of dynamic memory, leaving 2151 bytes for local variables. Maximum is 8192 bytes.

FastLED 3.9.17
Sketch uses 96142 bytes (37%) of program storage space. Maximum is 253952 bytes.
Global variables use 6695 bytes (81%) of dynamic memory, leaving 1497 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.

sutaburosu, 2025-05-25:
> Have confirmed this on our Mega 2560 project as well.

@nomakewan, thanks for confirming the increase in RAM usage and binary size. Did you try flashing the resulting binary to a device, and does it work? I can't get very simple sketches to display anything on just a single LED. As [I mentioned in the release announcement on r/fastled](https://www.reddit.com/r/FastLED/comments/1kuv7rl/fastled_3917_released/mu4um17/), FastLED seems to get trapped in an infinite loop reallocating memory for a HashMap if FastLED.addLeds<> is present in the sketch.

nomakewan, 2025-05-26:
Apologies for the delay in responding; I did not originally attempt flashing to the device. I have done so just now and likewise confirm that the device is dead and no LED actions occur. 3.9.17 appears to be a death sentence for AVR.

sutaburosu, 2025-05-27:
Thanks to [a post on r/FastLED](https://www.reddit.com/r/FastLED/comments/1kwqysq/num_leds_set_to_an_odd_numbereverything_worksset/) I've just learned that FastLED 3.9.17 only works with an odd number of LEDs. Any even number of LEDs causes it freeze during startup.

nomakewan, 2025-05-27:
Interesting; that does track with my findings. The device in question defaults two two pins, one with 64 LEDs and the other with 62 LEDs.

Notably changing these so that they are odd does not reduce the massive bloat in program size and RAM, so even if it "works" with odd numbers of LEDs there is still clearly something far more fundamental gone wrong.

zackees, 2025-05-28:
Looks like a complex static object I used is being constructed unconditionally and not being tree-shooken out. Look like this is why it's being emitted. I see it in wowki.

I'm going to put the static inside a static function. That will defer initialization and also hopefully causes it to be removed by the linker since a function that is not called is side effect free.

zackees, 2025-05-28:
Okay, looks like this might be an issue related to Arduino IDE. PlatformIO is not showing this regression. The whole month the UNO platform has the exact same build size:

```csv
datetime,commit_hash,binary_size
2025-05-19T21:23:50-07:00,6e40a782f8d32dbaaa66d8e02c855b73358d9892,10662
2025-05-19T20:53:02-07:00,19ae075da2e8efb6735b7527636c6b65e2a04bef,10662
2025-05-19T19:32:08-07:00,dc25e80e0e31b97f91d36df14bd24aabc964917c,10662
2025-05-19T19:31:33-07:00,3aca40f16199ddca92e6ff9ce3738c0ff59d2afc,10662
2025-05-19T20:28:47-06:00,fd4863f7576cb688bc60176e9f760f2b4c0414fb,10662
2025-05-19T02:57:25-07:00,4d670929a3c3ab762e502542e729d28746e48ef0,10662
2025-05-18T13:37:43-07:00,3cccccfd88b5b1602e777a64e8884cf2f2e9a5ec,10662
2025-05-18T13:16:24-07:00,2953bfc1e6e0a53ac6dbd2846440cb7e8691be53,10662
2025-05-17T17:21:48-07:00,eb6fb6855add1a7dd5eb7caa904cad45985c6d71,10662
2025-05-16T14:58:07-07:00,11e03d5684af51f3acb6801f5f85be0e28ccbaba,10662
2025-05-16T14:41:31-07:00,a58623539ae55cf2846294b5b0bbeca21ad9c4d3,10662
2025-05-16T10:44:10-07:00,f3f6811350e9b78727bf110363500150c848dd3c,10662
2025-05-16T00:40:32-07:00,124719160f30a6e69f22f931c5b5e3953228e9c9,10662
2025-05-16T00:36:56-07:00,743ceeaa671bab48191946d988f14baaef9ee545,10662
2025-05-16T00:24:56-07:00,219e0e9895d03120a96c004ad655e74553cff3d5,10662
2025-05-15T13:08:54-07:00,77a786e2fd74757f3772d1471cdffcd2020dbc14,10662
2025-05-15T01:38:01-07:00,a5c016551a623b47980aa0ded1ad40d464b54e25,10662
2025-05-14T23:11:01-07:00,dd8b880acf8c82a36c9add57b607f3337ffb216e,10662
2025-05-14T21:04:14-07:00,e6ae4586350c25afdb46f00670e7396867110315,10662
2025-05-14T20:22:30-07:00,e2a7b045b690979eeb41b913ada38db29b5951b5,10662
2025-05-14T13:34:05-07:00,4c3009d3eebb2f6a1a0a53efc7e4406ddd0cbbb2,10662
2025-05-13T18:31:43-07:00,4491268a059363d8d43d06e2a5331c37d07235c8,10662
2025-05-13T18:05:32-07:00,913be758e7687eb4c4d780225cd2ff1841879576,10662
2025-05-13T17:23:19-07:00,0d3b23dcbc4f193075426d45de192c980fe1c232,10662
2025-05-13T17:21:46-07:00,b17ea8b0c6dd7c688cadb1616b8221f3c7eb0a0f,10662
2025-05-13T17:19:52-07:00,76c2b8fa4950bc568f3457ce0a5ba5497b5726dc,10662
2025-05-13T17:17:58-07:00,28e4c1101c4c96f17eabf30b1ca0b7dc45046404,10662
2025-05-13T17:14:45-07:00,3406a313a61ea9bd2420539969df6f796c035692,10662
2025-05-13T17:12:52-07:00,0047605f30e33a76dcc0e314e351d46ca9c3c79b,10662
2025-05-13T17:11:28-07:00,c476e55a4cc778f2c32a4d02ef475982d7b2fa6a,10662
2025-05-13T17:06:39-07:00,8b613f6da3379cbbd031b983cd5d2274d196acd2,10662
2025-05-13T15:22:40-07:00,0f52370246699e04370ef7dcab21a1dbd625d3c4,10662
2025-05-13T14:45:23-07:00,f1dbf042ab1316e8782f25e3d665084de68bba50,10662
2025-05-11T20:53:45-07:00,64d5183df01517e1519bbd738596ddf3b7f383e2,10662
2025-05-11T20:40:32-07:00,b342e86ec2c11524cfb1c025ce221cc581fa942a,10662
2025-05-11T20:26:52-07:00,4338e8f3bbc2983fda37bcf8223e81e25bf7c379,10662
2025-05-11T20:26:40-07:00,517b40982157c83cf25be08dac5d7238b570686e,10662
2025-05-11T16:27:05-07:00,4283016edcf75158b737ed66f1f68a63d5b34783,10662
2025-05-11T15:33:29-07:00,6f7f6b696d554e45ce49082731f1025993c478ee,10662
2025-05-11T15:30:00-07:00,d56a1480210f3da41034447f5e8d82299af2b7f0,10662
2025-05-11T11:13:45-07:00,ecad31344e0712fa9e295574d5e3139b2dfac213,10662
2025-05-11T05:48:37-07:00,a970cbccc99ceded80c97f2ea106495cb550e14e,10662
2025-05-11T05:32:55-07:00,b2e7f280db1e9b1d5a968a0f98b291b34ea5ee30,10662
2025-05-11T05:31:31-07:00,9bda2c71bcb1cb8da240d351459c1421bf4455ae,10662
2025-05-11T05:24:59-07:00,255a0ca89def9faf63cb3c724a1c4b1d21db8774,10662
2025-05-11T04:23:48-07:00,6c3aaf8336bf4c777d692536718c1c309286eca4,10662
2025-05-11T02:21:27-07:00,15d7ed5a4b2fc13d818d3a3e9cd793272f3111d0,10662
2025-05-11T02:14:49-07:00,de0998db3a424fd8049e8abb7726a47a6f4c3fc0,10662
2025-05-11T01:57:21-07:00,5cc08fad404eb4a35615930c6dbe2803b35f1796,10662
2025-05-11T01:36:19-07:00,f71c98132dcd6617c790a37a92f472dfc208f8b7,10662
2025-05-11T01:19:37-07:00,8d708cb5803b184bc11a9805ff39c5df1410bf47,10662
2025-05-11T00:29:27-07:00,1282a13d65d2115b6f26c92f0440f6d22b23013f,10662
2025-05-11T00:25:00-07:00,a91636441a055043deca13432a9fb7b761a8483c,10662
2025-05-10T23:40:36-07:00,8ecc1e863d3c611587268ad1e373b224cf9d6421,10662
2025-05-10T22:21:32-07:00,5c6d616b6103c0adbe94332983951ea830c50aa6,10662
2025-05-10T21:37:04-07:00,a371ad6ce29efee3c8ee62c0b968d0a4268b620f,10662
2025-05-10T20:20:46-07:00,c73ca7f34b9cefa86e93e804be169a2420ede27b,10662
2025-05-10T17:37:43-07:00,c68be9ac614fc8dec2da6927c0c88bc80dd2f5df,10662
2025-05-10T17:06:24-07:00,022eea3749a2bcfd5a0d7fcb4596d72bdbd38654,10662
2025-05-10T05:20:44-07:00,a3d21514b1d8e7b735ebb3549b820488ca0842e5,10662
2025-05-10T05:19:35-07:00,f03fb5d6b37c4fd2e6260ffd738fd3ca4ace71c4,10662
2025-05-10T03:57:17-07:00,9666cd9494ed63de128d7f08f21e435ef4aedfdc,10662
2025-05-10T03:55:52-07:00,a098796059b39147d48e104a5c431eb2f625cd72,10662
2025-05-10T03:29:25-07:00,0c25bbbc540754303b6058f363de87ec0d567c5b,10662
2025-05-10T00:34:42-07:00,cfb24129bf553721fc8588e5d98a7a1b8e5e6761,10662
2025-05-10T00:09:38-07:00,31ff134ece8adc5a8382f570b4bb6f0f07005cdc,10662
[... 59 lines cut ...]
2025-05-08T15:05:45-07:00,b7d7639610afe93329907f37375a2135ff1f460c,10662
2025-05-08T14:53:24-07:00,639a74ace3bdbe63555a2975ccb16e46d22b97a8,10662
2025-05-08T14:52:57-07:00,1e8ba614950affbef794388205f33934744079b7,10662
2025-05-08T14:42:59-07:00,748f02c0a8d869da73bcaec602e2705c5498bc31,10662
2025-05-08T14:38:25-07:00,6d83231e5fb5bf2bdd165654cc989a31466b1852,10662
2025-05-08T14:36:02-07:00,1efeefb8e0f36d57a2f2e90137695f650ca8c69a,10662
[... cut ...]
2025-05-08T08:21:42-07:00,c9159c6b6b314e31352aa3236abab546172870da,10662
2025-05-08T08:19:45-07:00,219ec2b0f46876fdee96640bb43133cb0c8ec1d5,10662
2025-05-08T08:16:55-07:00,152eda04a7fd0076e07bba5d0961fd7d78aae245,10662
2025-05-08T08:03:57-07:00,b7ff45d990747934cd5fdc9f4b1c5015600baa94,10662
2025-05-08T08:02:29-07:00,72c774f0cb90a5750e3597c7c8ea9b3497b407cf,10662
2025-05-08T08:01:43-07:00,c2b5c2f273c89fab4cf59a0b5086e6f51cd082ac,10662
2025-05-08T07:26:04-07:00,2de91b57f308de317feae9090e5262a9b8ca01b9,10662
```

nomakewan, 2025-05-28:
> Okay, looks like this might be an issue related to Arduino IDE. PlatformIO is not showing this regression. The whole month the UNO platform has the exact same build size:

Zach, we use PIO exclusively in our project and, as I posted above, you can see that the binary size and especially RAM usage have exploded on our Mega 2560 builds built with PIO. Further, the issue where the AVR is totally dead after flashing the binary compiled with PIO is still there.

This is not an issue with the Arduino IDE.

zackees, 2025-05-28:
> EDIT: Also the log you posted only goes back to 5/19, and 3.9.17 was released on the 24th. Possible that whatever introduced this behavior was done in those 5 days?

It got truncated because my script aborted whenever one of the builds didn't work. But otherwise its all the same number.

zackees, 2025-05-28:
I submitted a fix already. If you want to run it again and see if it's fixed. I'll begin taking your platformio.ini file and try to reproduce.

nomakewan, 2025-05-28:
> I submitted a fix already. If you want to run it again and see if it's fixed. I'll begin taking your platformio.ini file and try to reproduce.

Tested [dfa2b8c](https://github.com/FastLED/FastLED/commit/dfa2b8c9d2115ecacc186082f2c3800b13384422) against our project using the Arduino IDE. Here were the size results:

```
Sketch uses 96142 bytes (37%) of program storage space. Maximum is 253952 bytes.
Global variables use 6695 bytes (81%) of dynamic memory, leaving 1497 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.
```
This appears to be unchanged from 3.9.17. Is the fix in a separate branch from `main`?

zackees, 2025-05-28:
Can you help me understand why you are compiling a complex project defined by a platformio.ini file through the arduino IDE?

Do you have a parallel process where you install each of these dependencies through the Arduino IDE?

zackees, 2025-05-28:
> Tested [dfa2b8c](https://github.com/FastLED/FastLED/commit/dfa2b8c9d2115ecacc186082f2c3800b13384422) against our project using the Arduino IDE. Here were the size results:
>
> ```
> Sketch uses 96142 [...]

Oops I forgot to check it in, here's the patch:

https://github.com/FastLED/FastLED/commit/9696074c3a267ed4237a3caa63a2fd079a48c19d

zackees, 2025-05-28:
Okay making progress:

```
Before:

RAM:   [=         ]  12.0% (used 986 bytes from 8192 bytes)
Flash: [          ]   3.8% (used 9648 bytes from 253952 bytes)

New:

RAM:   [          ]   4.1% (used 339 bytes from 8192 bytes)
Flash: [          ]   3.3% (used 8392 bytes from 253952 bytes)
```

zackees, 2025-05-28:
I don't trust these numbers. It's now indicating it's less than 3.9.16

nomakewan, 2025-05-28:
> I have no idea what's going on but now master is showing that it's a lot smaller. Can you confirm?
> 
> ```
> Main:
> RAM:   [          ]   4.1% (used 339 bytes from 8192 bytes)
> Flash: [          ]   3.3% (used 8392 bytes from 253952 bytes)
> 
> 3.9.16
> RAM:   [=         ]  12.0% (used 986 bytes from 8192 bytes)
> Flash: [          ]   3.8% (used 9648 bytes from 253952 bytes)
> ```

Here is the result for our project using PIO (sorry, had to look up targeting to a git commit):

`master`:
```
RAM:   [=======   ]  73.0% (used 5979 bytes from 8192 bytes)
Flash: [====      ]  37.3% (used 94658 bytes from 253952 bytes)
```

`3.9.17`:
```
RAM:   [========  ]  80.9% (used 6626 bytes from 8192 bytes)
Flash: [====      ]  37.7% (used 95816 bytes from 253952 bytes)
```

`3.9.16`:
```
RAM:   [=======   ]  72.9% (used 5972 bytes from 8192 bytes)
Flash: [====      ]  36.3% (used 92070 bytes from 253952 bytes)
```

I cannot confirm the smaller filesize on `master` with our project.

Also, I just tested our project by setting a single pin to have 1 LED more than previous and the board fired right up, further corroborating what @sutaburosu mentioned earlier. If the total number of LEDs initialized is odd, the board runs. If it is even, it will fail to leave `setup()`. Note that it appears to be the `total` number of initialized LEDs; setting two pins to an odd number results in an even total, which fails to exit `setup`. Changing a single pin to have 65 LEDs instead of 64 and everything seems to be working fine.

If you'd like to poke at the project it's here: https://github.com/gpstar81/GPStar-proton-pack/tree/v6.1.Dev/source/ProtonPack

It has 64 LEDs on pin D53 and 62 LEDs on pin D13. All LEDs are NeoPixel. The pack code has a POST sequence that tests all attached LEDs upon initial boot meaning as long as you have some LEDs connected to those pins, they should respond the moment the code starts running.

nomakewan, 2025-05-28:
> Can you help me binary search at all?
> 
> Does this break for you or does it pass?
> 
> 2025-05-13:
> 
> https://github.com/FastLED/FastLED#8b613f6da3379cbbd031b983cd5d2274d196acd2

```
RAM:   [========  ]  80.9% (used 6626 bytes from 8192 bytes)
Flash: [====      ]  37.7% (used 95816 bytes from 253952 bytes)
```
Has the exploded flash/RAM, and fails to boot when total LED count is even, same as 3.9.17.

nomakewan, 2025-05-28:
> Great, we are getting closer:
> 
> 2025-05-08:
> 
> https://github.com/FastLED/FastLED#b64551c02075ae0c7e9b15a8a5aef6dac1fdaf72

_She's aliiiiiiiiiiiiiive!_

```
RAM:   [=======   ]  72.9% (used 5975 bytes from 8192 bytes)
Flash: [====      ]  37.3% (used 94662 bytes from 253952 bytes)
```

Boots up first try.

zackees, 2025-05-28:
Okay, now we have our upper and lower bound:

It broke between May-8th and May 13th:

Can you try this next?

2025-05-11

https://github.com/FastLED/FastLED#a91636441a055043deca13432a9fb7b761a8483c

zackees, 2025-05-28:
I'm going to step out and get dinner. If you keep on binary searching then we'll pin down the exact cl that broke this. I'm pretty good with micro commits so that issue should become glarily obvious.

nomakewan, 2025-05-28:
I actually tested two commits. The first one I tried was the last time that the teensy4.1 binary size test succeeded, which was commit 1d2b767ad9161a96935bd54d786caa8cbef3a23d. That resulted in identical resource results to 3.9.16 and a functioning AVR:

```
RAM:   [=======   ]  72.9% (used 5972 bytes from 8192 bytes)
Flash: [====      ]  36.3% (used 92070 bytes from 253952 bytes)
```

I also tested your requested commit and had the bloated resource results and a dead AVR:

```
RAM:   [========  ]  80.9% (used 6626 bytes from 8192 bytes)
Flash: [====      ]  37.7% (used 95816 bytes from 253952 bytes)
```

zackees, 2025-05-28:
We have two issues here, the blowup and the odd even issue.

Blowup has been identified and fixed.

Now we just need to identify the CL that broke wrt even / odd leds.

Can you continue to binary search?

nomakewan, 2025-05-28:
Commit 8121c6b7ee65356c851b53fddc7dbb31121749d8 is broken.

Testing commit 0c14eeb75dc656145dccd6a4eb3e22236762120d now.

zackees, 2025-05-28:
Okay just to confirm, the use case is as simple as the blink sketch? And setting leds to 2 works but just 1 doesn't.

Nothing in this CL makes any sense.

So i'm just going to start disabling code for the new fancy stuff.

zackees, 2025-05-28:
How much longer are you going to be up? I can just start disabling code in 2 or three cla to break this problem up.

zackees, 2025-05-28:
Okay then we'll use your project as a test case. I'm back at my computer now and will continue.

nomakewan, 2025-05-28:
> I made a branch for you, please confirm whether this problem is fixed in this branch.
> 
> https://github.com/FastLED/FastLED#fix-odd-leds

Can confirm that this branch does not crash. Here was the resource results:

```
RAM:   [=======   ]  73.0% (used 5979 bytes from 8192 bytes)
Flash: [====      ]  37.3% (used 94658 bytes from 253952 bytes)
```

zackees, 2025-05-28:
FastLED 3.9.18 release candidate:

Does this commit solve BOTH the memory issue and the odd/even led issue?

https://github.com/FastLED/FastLED#07a759cdfab4ebd342c3b15be6dbdaa778e93712

zackees, 2025-05-28:
Okay, I can find out where the 600 bytes are coming in from later, through the auto generated symbol maps on our builders. I can do a size comparison on what got leaked in.

Thanks for your help! I merge in this change it will be submitted tonight.

zackees, 2025-05-28:
https://www.reddit.com/r/FastLED/comments/1kxalk1/3918_hot_fix_is_out/

sutaburosu, 2025-05-28:
I just ran `git bisect` from bc2a1d874996af0401b81a812ca401607580ef96 -> HEAD. I built each revision twice with odd/even number of LEDs in the Arduino IDE. I uploaded both the HEXes to Wokwi to see the results. That also landed on 595e2799a516843ed127d0c53a37ff227bd31660 as the first bad commit.

zackees, 2025-05-28:
Round 2

sutaburosu, 2025-05-28:
I don't pretend to fully understand the problem, but searching gives me impression that anonymous namespaces are the modern alternative to static globals. Perhaps even the ancient GCC 7.3.0 understands them and might do the right thing in this case?

zackees, 2025-05-28:
I think the issue is that modern gcc removes globals if they think no one is linking to them.

If the constructor is meant to invoke the desired side effect, then the result is that the constructor doesn't get run. This is surprising until the first time you experience it and to get around it you add a gcc attribute.

It's not a good compiler feature and i believe it's theoretically wrong. However, I'm sure they get so much code removal because of it that they keep it in anyway and force people to mark the constructors as keep alive.

But the older compiler's i don't think did this, and I think that's what we are having happen now. Older gcc keeps the constructors, new gcc/clang elides them.

The fix is easy, just stick the global as a static variable inside a static function. Static constructors inside a function don't get run until the first call. And if no one links to that function, it can be removed.

zackees, 2025-05-28:
Does this commit solve your even odd issue?

`https://github.com/FastLED/FastLED#0f9da866a6e096347f3a454d1a375d062fb88acd`

sutaburosu, 2025-05-28:
Yes, yes it does! It also brings down the runtime memory usage to the same as 3.9.16.

The binary size is also basically the same as 3.9.16, with just a couple of extra bytes used when only #include'ing, but gained back when actually using FastLED.

Awesome. Thanks, Zach!

```
FastLED 3.9.16 | 223 (+17)    | 306 (+83)       | 334 (+28)     | 209/292/320    | 1946/4314/5234
FastLED 3.9.17 | 1338 (+1134) | 1421 (+83)      | 1449 (+28)    | 869/952/980    | 6424/8792/9712
FastLED 3.9.18 | ? hangs      | 766 (+??)       | 794 (+28)     | 222/305/333    | 5168/7536/8456
FastLED 0f9da86| 223 (+17)    | 306 (+83)       | 334 (+28)     | 209/292/320    | 1948/4314/5234
```