FastLED/FastLED #1938: Variable orientation in FireMatrix?

4wheeljive, 2025-06-08:
Before getting to my question, I want to give a big shout-out to u/zackees. In addition to everything else he's been working on, I noticed yesterday that he has added some detailed and incredibly helpful step-by-step comments in several of the examples (e.g., FireMatrix, FxWave2d). **I can't tell you how valuable this will be for noobs like me in figuring out how all this stuff works and adapting it for use in our own projects. Thank you, thank you!!**

Now...I have a question about the use of WIDTH and HEIGHT in various parts of the FireMatrix example.

My understanding is that the standard order/structure for XYMap class functions is (x, y) and (width, height) (i.e. horizontal first, then vertical).

But the FireMatrix XYMap at line 86 puts HEIGHT first, then WIDTH: 
`  XYMap xyMap(HEIGHT, WIDTH, SERPENTINE);
`

Also, throughout most of the sketch, i and j are used as the horizontal and vertical coordinates respectively. But lines 163 and 164 diverge from this: 
 ``` 
for (int i = 0; i < HEIGHT; i++) {
     for (int j = 0; j < WIDTH; j++)
 ```
Followed by line 175:
  ` int index = xyMap((HEIGHT - 1) - i, (WIDTH - 1) - j);`

I'm guessing that all of the above cancel each other out, and there wouldn't be any issues running this sketch on a square matrix (such as 100x100 in the example). But I had trouble adapting the code to my 48x32 panel with a custom XYMap, until I noticed and reversed the "nonstandard/swapped" variables above. My initial thought was that the swapping was inadvertent, but I started second-guessing that when I re-read the comments starting at 172:  
```
// Convert our 2D coordinates (i,j) to the 1D array index
// We use (HEIGHT-1)-i and (WIDTH-1)-j to flip the coordinates
// This makes the fire appear to rise from the bottom
int index = xyMap((HEIGHT - 1) - i, (WIDTH - 1) - j);
```
At first I thought this merely explained the conversion from 2D to 1D. But the mention of a coordinate "flip" gave me pause. My tired brain couldn't immediately process whether this meant just a vertical flip, or if the logic required a 90 degree rotation and was the reason for the swapping.

(The original sketch (https://pastebin.com/jSSVSRi6) didn't involve any rotation, so I don't think that's what's intended here. But with all of the XYMap magic going on in the background, I'm not so sure.) 

Is there something going on here I'm not understanding (always a distinct possibility!), or might some tweaks in the example sketch perhaps be order?

4wheeljive, 2025-06-08:
I tried making some suggested edits on a forked copy ( https://github.com/FastLED/FastLED/blob/d6ef30ba580509f961ca47990a9653a84401ee84/examples/FireMatrix/FireMatrix.ino ) to potentially submit a PR, but there's something unusual happening.

My changes look fine on a square matrix. If I make the matrix wider, than taller, it looks sort of okay, but it shows only the top portion of the fire. If I make the matrix taller than wider, the whole fire is elevated and appears to burn both up and down!

With the HEIGHT and WIDTH variables as they were, the fire seems properly positioned vertically, but the orientation of the screenmap display does not match what the code would suggest (i.e., when WIDTH is 48 and HEIGHT is 32 -- which should appear as "landscape" -- screenmap displays it as "portrait").

So, I think that to get the sketch to work reliably for users in with variety of matrix orientations, two things need to happen:
- HEIGHT and WIDTH need to be swapped in the several places I pointed out
- Some other offset/adjustment needs to be made to ensure an appropriate vertical positioning of the fire, whatever the layout

I'll see can if I can come up with something for the second tweak.

4wheeljive, 2025-06-09:
Okay, solving the second item I mentioned above turned out to be quite simple. I just swapped HEIGHT for WIDTH in the subtraction_factor function as well. 

I didn't touch that previously, as I thought that having the vertical subtraction_factor be a function of the horizontal position might be part of what creates a realistic fire effect. But as I thought about it more, I decided this wasn't the case and that it was probably the source of my vertical alignment problem.

zackees, 2025-06-09:
Yeah I think you found a mistake on my part. The height and width thing were likely because accessing a c-array is column major rather (y first) instead of row major(x first), the latter being intuitive to programmers.

I assume you are using the web compiler so iteration on your part should be fast.

Is it too much to ask if you can correct this issue since you are in there anyway?

Let me know. No is ok too.

\~Z~

4wheeljive, 2025-06-09:
All that was needed in Fire Cylinder was swapping HEIGHT and WIDTH when xyMap was created. You already straightened out everything else when adjusting for the extra dimension, and the xyMap issue wasn't noticeable because the matrix was square. I submitted a PR.

closed 2025-06-09 by 4wheeljive

zackees, 2025-06-09:
I noticed that the colors are washed out using the default graphics driver due to the intensity of the light and the bloom effect i'm using.

I've gone ahead and fixed this by converting the XYMap to a fl::ScreenMap manually and then setting the led diameter to 1mm (.1). Eventually I plan on fixing this automatically with a dynamic led diameter that will simulate the eyes adjusting the iris due to global average light intensity.

Note that you can force the more classical graphics rendering driver by appending `?gfx=0` to the url. By default it's using `?gfx=1` implicitly.