FastLED/FastLED #1960: Noise Functions causing crashes on ESP8266

AlbertGBarber, 2025-07-02:
Hi, I'm running the latest versions of FastLED (3.10.1) and the Arduino ESP8266 Core (3.1.2) on a Wemos D1 Mini V3.0.0, and it seems like the latest FastLED hotfix is causing the ESP to crash when calling `noise8()` or `noise16()` (haven't tested any others).

My code is all wrapped up in my [Pixel Spork](https://github.com/AlbertGBarber/PixelSpork/wiki) library so I can't really post it fully here, but I have a number of effects that use noise8 or noise16, and all the ones I've tested cause my ESP to crash as soon as the noise functions are called.

For example, just calling `noise8()` with fixed inputs like: `noiseData = inoise8(50, 40);` causes the ESP to crash (`noiseData` is a uint8_t). I've isolated the crash to the noise functions, as removing them (ie setting `noiseData` to a fixed value) stops the crashing. 

Everything was working fine as of FastLED version 3.10.0. I haven't tested the FastLED noise examples, as I haven't had time to create a setup for them. 

I apologize in advance if this is an issue on my end! My library is pretty complex, and I've combed over everything I could think that may cause the problem -- I don't take creating an issue here lightly as I know it creates work for you!

mcspr, 2025-07-03:
Message above is not a crash, just a normal bootloader output.

I've ran the sketch in question w/ 3.10.1, and see that noise.cpp tries to access `p[]` which is located in flash b/c of `FL_PROGMEM` attribute
```
Exception (3) - LoadStoreError: Processor internal physical address or data error during load or store
epc1=0x40209bd9: inoise8_raw(unsigned short, unsigned short) at /home/runner/Arduino/libraries/FastLED/src/noise.cpp:605
excvaddr=0x402427fa: system_get_sdk_version at ??:?
sp: 3ffffdf0 end: 3fffffd0 offset: 0150

ctx: cont
0x40209dd7: inoise8(unsigned short, unsigned short) at /home/runner/Arduino/libraries/FastLED/src/noise.cpp:637
0x402035d4: LavaPS::update() at /home/runner/Arduino/libraries/PixelSpork/src/Effects/Lava/LavaPS.cpp:85
0x3ffee780: ?? at /home/runner/dev/LedSpiral/LedSpiral.ino:72
0x3ffee7d4: ?? at /home/runner/dev/LedSpiral/LedSpiral.ino:69
0x40208c58: EffectSetPS::update() at /home/runner/Arduino/libraries/PixelSpork/src/UtilEffects/EffectSet/EffectSetPS.cpp:128 (discriminator 2)
0x4020a784: loop_wrapper() at /home/runner/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp:264
0x40100d21: cont_wrapper at /home/runner/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S:92
```

ESP8266 Core historically uses AVR terminology for flash access, which always requires 32bit aligned access to work. We do provide both `PROGMEM` & `pgm_read_...` macros for the same reason.

I would guess that adding `#define FASTLED_USE_PROGMEM 1` before `#include <FastLED.h>` should help the issue, because when building we end up with progmem disabled but `p[]` is using the flash memory attribute regardless
https://github.com/FastLED/FastLED/blob/dec101a1268e013ef42deee644ff58883179d61a/src/platforms/null_progmem.h#L8

Current dev branch uses quite a different style of that header, so maybe it is already fixed.

zackees, 2025-10-08:
FL_PROGMEM was fixed in 3.10.3, closing as fixed.

closed 2025-10-08 by zackees