FastLED/FastLED #1961: I2S and WiFi are not compatible (ESP32-S3)

clevertoys, 2025-07-07:
Maybe this is simply a platform limitation and not solveable, but in any case I have tested the combination of I2S clockless parallel driver with and without the WiFi stack running and can say with some degree of confidence (I could be missing some trick to get this to work, but so far have not found any ) that the two subsystems are incompatible. My guess is that WiFi stack generates a ton of interrupts and this disturbs the clock for the WS2812 protocol output on I2S. I have found an interim solution for my project which is to reduce the required physical pins/strips to 4, and use the RMT driver. RMT and WiFi seem to get along great :)

zackees, 2025-07-07:
Yeah this is known problem. RMT 4 has lots of low level code to make this work correctly. I thought the I2S was robust because of the DMA but apparently it doesn't work as well as I thought, as you are showing.

It might work better if you break up your packets to smaller sizes. I suspect that large frames being sent just drive all the interrupts in the wifi to max.

clevertoys, 2025-07-07:
No worries. Maybe there should be a definitive guide to the I2S and RMT drivers, their pros/cons... Closing...

clevertoys, 2025-07-08:
Sure. No problem.

zackees, 2025-07-20:
So I've got an update. From the reddit history it looks like I2S and Wifi were compatible at some point in time.

Better than vegas odds if you downgrade your ESP arduino core back 4.4 idf esp, both rmt and I2S will work better with wifi.

You should know that 5.0 broke a lot of people. 5.2,5.3 idf straight up broke I2S, and then it worked again in 5.4

All the led product like WLED are esp idf 4.4 and that's because despite being old, it was a superior api for clockless stability.

So given all this, i really think if you downgrade your core... it's going to work. And if it does, will you please let me know so I can update the readme