FastLED/FastLED #1963: Request for proper documentation of functions and namespaces.

memeticae, 2025-07-09:
I have just spend the better part of a week trying to get "fill_gradient_RGB()" to work again, after updating from 3.9.16 to anything newer.

As I finally discovered, the fill_gradient_RGB() function "suddenly" belongs to the namespace "fl". 
Now, when using fl::fill_gradient_RGB(), my code works with *both* the new /and/ the old libraries. 
Which I find odd.

After reviewing the source, it seems a bit haphazard which functions belong to the namespace "fl" and which do not, yet have duplicates, or not.

If the purpose is to use C++, then use C++, but please! Do document which functions reside in which namespace.

Thank you.

zackees, 2025-07-24:
Everything is moving to the fl namespace with legacy code bindings bringing them into the global namespace.

Sorry you hit a corner case. I'll correct.

closed 2025-07-25 by zackees

zackees, 2025-08-01:
Documentation will be coming later. I prioritize examples as documentation.