FastLED/FastLED #1965: Issue with the 1284P and FastLED versions newer than 3.1.0 - Dark strip

SteveH01, 2025-07-13:
Hello!

I have an old project that drives leds around the outside of my Bass amp.  It has been running fine for over 8 years on the 3.1.0 library.  I'm using a 1284P chip (mighty1284P in the Arduino IDE).  I'm trying to update my code and am noticing that while my sketch compiles with higher releases of the library, the LEDs on the strip remain dark.  I can revert back to 3.1.0 and things start working again.  I also see this behavior on the Demo100.ino sketch.  I've also tried using the #define "FASTLED_ALLOW_INTERRUPTS 0" directive with no change in results.

What do I need to do to get the 1284P working with newer versions of the Library?

Thanks!

Steve Hurd
Floyds Knobs, IN

zackees, 2025-07-24:
Hey steve, can you try some other version of the code. You have 3.1 which is ancient.

What about 3.3, 3.6, 3.9?

SteveH01, 2025-07-24:
> The 1284p is a rugged chip from a more civilized era.
>
> After careful deliberation, sadly, I'm not willing to break into the local
> museum to get a test unit to support your request.
>
> May I suggest looking into an esp32dev? This is the 🐐 of embedded led
> development now.

The problem is I have an old enclosure and a LOT of home-grown Arduino
sweat and tears surrounding this chip.

[image: image.png]
[image: image.png]

Is the 1284P officially no longer supported?

I've got no problem going with the newer chip, but I hadn't planned on a
full redesign of the hardware at this point.

When did the 1284P fall out of favor?  I have a 3D printer with the same
chip that's only about 5 years old.

Thanks,

Steve Hurd

SteveH01, 2025-07-25:
> Oh I'm wrong, it's a beefy 328p but at 20mhz. Most likely the timings are
> wrong. We have a FASTLED_OVERCLOCK factor but it doesn't apply to avr
> because avr has it's timing in rsolution of 100ns so it has not overclock
> factor applied. But it probably could and that might be able to fix the
> issue for you.

Had to dig for the pic, but here's what I have.  It's a custom breakout
board I ordered from someone in the UK in 2016.  It was the only one I
could find at the time.  I went through several boards looking for one that
was beefy enough for my code/features.

[image: image.png]

What do I need to do to implement the overclock factor?

Steve Hurd

zackees, 2025-07-25:
[src/chipsets.h](https://github.com/FastLED/FastLED/blob/25b9961d47236691c664e901ed022162c2c02637/src/chipsets.h#L918)

zackees, 2025-07-25:
My guy is that your `CPU_F` is off

https://github.com/FastLED/FastLED/blob/25b9961d47236691c664e901ed022162c2c02637/src/chipsets.h#L829