FastLED/FastLED #1966: using FASTLED_USES_ESP32S3_I2S, color order not set by RGB_ORDER?

ewowi, 2025-07-13:
Testing I2S drivers on S3 using IDF 5: [platform 54.03.20 ](https://github.com/pioarduino/platform-espressif32/releases/tag/54.03.20)

It looks like the RGB_ORDER part of the FastLED.addLeds is not used to set the I2S driver.

See [driver.h](https://github.com/FastLED/FastLED/blob/master/src/third_party/yves/I2SClockLessLedDriveresp32s3/driver.h) which sets COLOR_ORDER_RBG

See [I2SClockLessLedDriveresp32s3 .h](https://github.com/FastLED/FastLED/blob/master/src/third_party/yves/I2SClockLessLedDriveresp32s3/src/I2SClockLessLedDriveresp32s3.h) which uses this to define _p_r, _p_g, _p_b

So RBG looks to be hard coded.

Is it possible to use RGB_ORDER to set the right color ordering?

ewowi, 2025-07-13:
In the non-s3 I2S driver colorarrangement is used as a parameter of initLed (the addLeds of I2S drivers). That could be used for inspiration : [initled](https://github.com/ewowi/I2SClocklessLedDriver/blob/201d866f56a9c18fc84654d5a1f552873ad02a1d/src/I2SClocklessLedDriver.h#L998-L1054)

cc @hpwit

ewowi, 2025-07-14:
Also ML_CHIPSET in 

FastLED.addLeds<ML_CHIPSET, 40 COLOR_ORDER_ARG>(leds, startLed, nrOfLights)

Seems to be irrelevant. I2S driver supports only WS28.. and SK6812 without needing to specify it (https://github.com/hpwit/I2SClockLessLedDriveresp32s3)

zackees, 2025-07-24:
Weird since other people have said this too.

The color should not be sent to the driver itself, the PixelController<COLOR_ORDER> that is sent to the driver is a defacto pixel iterator that does inline color re-ordering on the bytes as it writes out to the driver.

zackees, 2025-07-29:
I'll check it

zackees, 2025-07-29:
Can you give a few words on the newer driver?

Does it allow printfs while in use? Can the device be reflashed without hitting the reset button?

closed 2025-10-08 by zackees