FastLED/FastLED #1968: Fix for jumps/quantization of APA102HD mode

gwgill, 2025-07-17:
I was dissatisfied with a gamma corrected ramp using the APA102HD mode (it seemed to visibly step at one or more points), and some analysis of the 8 bit rgb + 5 bit brightness function showed why:
<https://www.argyllcms.com/APA102_loglog.svg>.

So here is a suggested replacement for five_bit_bitshift() in fl/five_bit_hd_gamma.cpp:

`// Improved FastLED five_bit_bitshift() function
// This appears to exactly match the floating point reference,
// with a worst case resulting error of 0.2% over the full 16 bit input range,
// and an average error of 0.05% over that range. Errors scale with maximum
// magnitude.`

zackees, 2025-07-19:
The return value was used for testing.

closed 2025-07-19 by zackees

zackees, 2025-07-24:
Interesting, and since it's pre-computed this would work fine.

Btw whats your math background?

gwgill, 2025-07-24:
My maths background isn't especially strong, but I learn what I need to know when I'm trying to do Electronic Engineering or Color Science, and an EE degree gives me a base.

zackees, 2025-07-24:
What was the process you used to come up with this arcane math? Was it numerically determined? It's like pure bit twiddling magic and I'm not quite how you did it.

gwgill, 2025-07-25:
The 5 bit brightness is a pretty straightforward idea :- it is analogous to a conversion to floating point format, except that the exponent (brightness) is a multiplier rather than a power of 2. So the idea is simply to maximize the number of significant digits in the 8 bit RGB values so that the largest is as large as possible without exceeding 255.  Maybe this helps:

```
/* Reference implementation */
/* 16 bit linear input, 5/8/8/8 bit output */
void ref(int out_rgbs8[4], int in_rgb16[3] ) {
    int i;
    double rgb[3];
    double rgbs[4];
    double mx = 0.0;

    /* Convert 16 bit input to fp */
    for (i = 0; i < 3; i++) {
        rgb[i] = in_rgb16[i]/(double)0xffff;
        if (mx < rgb[i])
            mx = rgb[i];
    }

    if (mx == 0.0) {
        for (i = 0; i < 4; i++)
            out_rgbs8[i] = 0;
    } else {
        /* Compute 5 bit quantized scale that is at or below maximum */
        rgbs[3] = ceil(mx * 31.0) / 31.0;

        /* Adjust rgb values for scale and quantize to 8 bits */
        for (i = 0; i < 3; i++)
            rgbs[i] = floor((rgb[i] / rgbs[3]) * 255.0 + 0.5) / 255.0;

        /* Convert back to 8/5 bit values */
        for (i = 0; i < 3; i++)
            out_rgbs8[i] = rgbs[i] * 255.0;
        out_rgbs8[3] = rgbs[3] * 31.0;
    }
}
```
So the conversion to integer maths without divides was to do the scaling in 32 bit and then shift divide back to 16 bit. Since the multiplier only has 32 possible values, a table to pre-compute the 32 bit fixed point multipliers was a logical move. The rest is a lot of fudging to account for the different scaling values of the integer/fixed point encodings, and trying to make things round the right way. Add in a bunch of trial and error tweaking of all the fudges.

Computing the 5 bit scale value is sort of the inverse of the idea of "how do you convert from 8 bit to 16 bit ?" to which the answer is 16bit = 8bit << 8 + 8bit, thereby closely approximating the scale change between 255 and 65535. It was all a bit of trial and error, and I'm guessing only works well in this case because the resolution is so low (5 bits), so that any inaccuracies are quantized out. 

The linear segment + power curve is a result of solving the two equations that define the resulting curve.

The curve part is: y = (a * x + b)^g with the crossover point d and the straight line slope c.
Because 1.0 has to map to 1.0, b always has to be 1 - a.
So we end up with two equations and two variables:
Match the slopes at the crossover point by making the differential of the curve equal the straight line slope:
c = a * g * (a * d + 1 - a)^(g-1)
and make the actual values match at the crossover point:
c * d = (a * d + 1 - a)^g
As I said, I couldn't find a closed solution, even throwing Maxsyma/wxMaxima at it, and in the end I uses wxMaxima to re-arrange the two equations so that I could iterate the solution. Since one equation is a differential of the other, I suspect the result was a Newton-Raphson method type iteration, hence the rapid convergence (See the Numerical Recipes in C section "Newton-Raphson Method Using Derivative"). 

Hope this helps.