FastLED/FastLED #1971: Trouble with new FL Json.ino

4wheeljive, 2025-07-23:
I was excited last night to discover the brand new fl::Json class source and example files. I'm not sure if the json.ino example was supposed to be ready for use yet, but it seemed like it was, so I tried testing it out.  I ran into several problems.
- fl::CRGB initialization at line 18
- Attempt to access a JSON doc array at line 78 (which seems to be looking back to FLArduinoJson, which I'm not sure it's supposed to be doing at this point).

Here's a screenshot of the problem list (would love to know if there's a better way to capture and share this info):
[image]

Here's the terminal log from a build attempt:
https://gist.github.com/4wheeljive/6384af16b1c2eda76a7619610ca1710e 

I tried troubleshooting for about 2 hours without success.

I look forward to seeing/using this once I can get it working! Thanks.

4wheeljive, 2025-07-23:
I just realized something regarding the line number references in my message above and in the logs. In trying to build the sketch in platformio, I had to move all of the core functions (i.e., everything after loop()) to before setup(). I dropped them in right after 

```
Forward declaration
void checkJsonType(const fl::Json& value, const char* fieldName);```

My reference to line 78 corresponds to line 108 in the actual example file.

zackees, 2025-07-24:
The new json stuff is not ready yet. I'm moving everything to dynamic dispatch so the header payload shrinks and I can massively improve rebuild time. That Arduino json is a massive header file of template magic that is broken in surprising ways. I've extracted the minimal api, making it easy, and pouring concreted on the bindings to FLArduinoJson in a cpp file.

Most of the lib has been ported except the JSON UI which is exceedingly stubborn.

However once this migration is complete FastLED will have the most advanced and fluid json api for embedded. Glad you found it!

4wheeljive, 2025-07-24:
Cool. Thanks for the update. I saw that you spent much of yesterday trying to straighten this out!

> The new json stuff is not ready yet.

> As of yesterday it should be much better and working.

Not sure how to reconcile these two statements. I just tested the sketch and am still getting the same kinds of errors (and a few more). I figure I'll just wait a bit more before testing out the new json stuff.

In the meantime, any idea why `fl::CRGB leds[NUM_LEDS]` is not working? It seems to me that this is a separate issue, which I also tried to troubleshoot, to no avail.

zackees, 2025-07-25:
Json write semantics aren't ideal. Tomorrow it will be done.

4wheeljive, 2025-07-25:
I’ll be happy to prove additional feedback on the code base. Thanks for asking. Let me think about what might be most useful.

closed 2025-07-28 by zackees