FastLED/FastLED #1974: compilation warnings

memeticae, 2025-07-31:
Why do I get these?

c:\Users\Documents\Arduino\libraries\FastLED\src\fl\screenmap.cpp: In static member function 'static bool fl::ScreenMap::ParseJson(const char*, fl::FixedMap<fl::Str, fl::ScreenMap, 16>*, fl::Str*)':
c:\Users\Documents\Arduino\libraries\FastLED\src\fl\screenmap.cpp:51:39: warning: unused parameter 'jsonStrScreenMap' [-Wunused-parameter]
 bool ScreenMap::ParseJson(const char *jsonStrScreenMap,

nomakewan, 2025-07-31:
You get those because of exactly what the warning says--because there are several functions within FastLED's codebase which declare parameters that are never used. Let's take a look at the one you provided as an example.

https://github.com/FastLED/FastLED/blob/3.10.1/src/fl/screenmap.cpp#L51-L57

Here we assume, since you are compiling `Blink.ino`, that you did not explicitly enable FastLED's JSON parsing. In this case, the branch of code that would be followed is this shorter path, and in this shorter path there are no references to the `jsonStrScreenMap` parameter. As a result, the compiler will throw a warning about the unused parameter.

This is just a warning, however, as the functions you are using within `Blink.ino` don't actually call `ParseJson` and so would not be linked in anyway.

nomakewan, 2025-07-31:
For that it would be best to direct the question to the maintainer; I can only explain why the warning occurs, not why it is present in the first place.

zackees, 2025-08-02:
Are you using a custom build or Arduiono/Platformio?

All these warnings are going to go away. Usually these platforms are running with -fpermissive and this shouldn't be an issue.

Internally the library is made with strict compiler settings, but things are in flux right now. They warnings will likely be gone by the next release.

closed 2025-08-02 by zackees

memeticae, 2025-08-02:
I have some more information.

"Optimize for debugging" was enabled in the arduino IDE (2.3.6)

When is disabled this a new warning arose:
```
C:\Users\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.1\cores\MCUdude_corefiles/Arduino.h:345:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   return (x < l) ? l : (x > h) ? h : x;
```

Since this seems to be a minicore message, should I raise the question there?

memeticae, 2025-08-02:
Thank you for taking the time, once I turned off "Optimize for debugging" all the warnings went away.

For some reason "Optimize for debugging" is set globally, instead of on a per project basis.
(I have already "issued" this elsewhere.)

zackees, 2025-08-02:
The error comes from a cpp file, that cpp file should be compiled once initially then the compilation artifact will be re-used.

I'd expect you to get one warning and then it goes away on subsequent compiles.