FastLED/FastLED #2000: Timing is off on Atmega328 when using optimization flag -Og

felias-fogg, 2025-08-02:
When the library is compiled for an ATmega328P with the optimization option "-Og" (debug-friendly), which you need for debugging your sketch (see [MiniCore](https://github.com/MCUdude/MiniCore)), then timing is off. Instead, the frequency is 500 kHz instead of 800 kHz, and all pulses have a width of 500ns (instead of 250ns and 860ns). I thought that an easy cure would be to force one of the modules to be compiled with -Os (with a GCC pragma). However, I have not been successful. 

Do you have a good idea how to mitigate the problem?

felias-fogg, 2025-08-03:
I looked a little deeper into the issue. It turns out that with -Og, the inline assembly code generated as part of the `showRGBInternal` method in `clockless_trinket.h` contains a number of (stupid) redundant move operations. It seems to be the case that for each asm statement, r10 is loaded into r16 and at the end this is reversed.  So, of course, the timing is off. 

Changing one line in the `ASM_VARS` macro from `[b1] "+a" (b1)` to `[b1] "=&a" (b1)` solves the problem. The code for the inline assembly code looks good when -Og is enabled, and for the standard optimization everything looks fine, the code is 100 bytes longer though. I must admit that I do not have an idea why this happening. 

All in all, this all looks pretty much like a compiler glitch to me.

zackees, 2025-08-04:
This is ancient code. Those redundant moves may expand out the timing in -Os so it works. And now in a debug setting the timing is off.

Obviously I don't want to touch the release version of this ASM code, it's pretty off limits since my asm chops are amature. However, I would accept a patch for a debug if-def to do the right thing as that is low risk.

zackees, 2025-08-05:
I understand, it's that every feature needs to be overridden manually if something like this goes wrong.

What you've selected is the right default. I have to deal with users where the default does might not work. If that's the case, I'll be the support guy, but you'll likely not be around.

So this is my only chance to have a conversation with you while you are contributing. After this I'm the support guy.

Like I said, I can reformat it and make a separate CL but I want to give you first offer of refusal to do the correct two phase macro that allows the default to be disabled. If the answer is "downgrade" or "wait till the next release" then that's a huge bummer.

zackees, 2025-08-05:
Fixed with https://github.com/FastLED/FastLED/pull/2021

felias-fogg, 2025-08-06:
Hi @zackees:

After having looked at the assembly code generation, I think it should be possible to do it in a less brittle way. By that, I mean that it should be possible to generate the assembly code in a way that is independent of the actual compiler optimization. 

Do you think it is worthwhile?