FastLED/FastLED #2060: ESP32-S2 I2S Linking Error

Seitekful, 2025-08-17:
Hi, 
when adding `#define FASTLED_ESP32_I2S true` I cant build the project anymore -> error when linking.
I am aware of these very similar issues [1810](https://github.com/FastLED/FastLED/issues/1810) & [1811](https://github.com/FastLED/FastLED/issues/1811) but was not able solve the issue... any advice?

platform.ini
```
platform = espressif32
; platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = lolin_s2_mini
framework = arduino
board_build.f_cpu = 80000000L
board_build.f_flash = 40000000L
board_build.flash_mode = dio
board_build.partitions = BuddyPartitions4MB.csv
board_upload.flash_size = 4MB
build_flags = 
	-D ARDUINO_USB_MODE=0
	-D ARDUINO_USB_CDC_ON_BOOT=0
	-D ARDUINO_USB_MSC_ON_BOOT=0
	-D ARDUINO_USB_DFU_ON_BOOT=0
	-D CORE_DEBUG_LEVEL=4
build_unflags = -DBOARD_HAS_PSRAM
lib_deps = 
	fastled/FastLED@^3.10.1
```

Build Output
```
Archiving .pio\build\V10\libFrameworkArduino.a
Linking .pio\build\V10\firmware.elf
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x0): undefined reference to `gCurBuffer'  
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x4): undefined reference to `dmaBuffers'  
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x14): undefined reference to `gPixelRow'  
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x18): undefined reference to `gDoneFilling'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x1c): undefined reference to `i2s_transpose_and_encode(int, unsigned long, unsigned long volatile*)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv]+0x4): undefined reference to `i2s_is_initialized()'      
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv]+0x8): undefined reference to `i2s_define_bit_patterns(int, int, int)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv]+0xc): undefined reference to `i2s_init(int)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv]+0x10): undefined reference to `i2s_set_fill_buffer_callback(void (*)())'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE4initEv]+0x14): undefined reference to `i2s_setup_pin(int, int)'  
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x4): undefined reference to `i2s_begin()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x8): undefined reference to `i2s_clear_dma_buffer(unsigned long*)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0xc): undefined reference to `i2s_start()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x10): undefined reference to `i2s_wait()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x14): undefined reference to `i2s_stop()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o: in function `ClocklessController<13, 250, 625, 375, (EOrder)66, 0, false, 5>::fillBuffer()':
C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/.pio/libdeps/V10/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:250:(.text._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv]+0x110): undefined reference to `i2s_transpose_and_encode(int, unsigned long, unsigned long volatile*)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/.pio/libdeps/V10/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:250:(.text._Z [... cut ...]
C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/.pio/libdeps/V10/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:292:(.text._ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi13ELi250ELi625ELi375EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x74): undefined reference to `i2s_stop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\V10\firmware.elf] Error 1
```

Seitekful, 2025-08-20:
Thanks for your reply, I tried to use `-DFASTLED_USES_ESP32S3_I2S=1` but it results in this error: `.pio/libdeps/V10/FastLED/src/platforms/esp/32/clockless_i2s_esp32s3.h:9:2: error: #error "This file is only for ESP32-S3"`

The ESP32-S2 has one I2S channel and I would like to use it - even if it means to change the library a bit. But I don't understand the build output, it seems to me that it cant find the files? Any clues how to tackle the linker issue?

zackees, 2025-08-20:
Your output shows that it's working correctly and can find all the files.

You can go ahead and remove the error and see if it works. It probably won't.

We have an alt driver for the esp32 called the spi driver. It works with just one pin. It's great for the c2, I'm assuming it works for the s2.

Use the FastLED master zip, as there is a fix in there for the driver that you'll need

Seitekful, 2025-08-20:
OK so removing the error has again resulted in an linking issue: 
'''

Archiving .pio\build\V10\libFrameworkArduino.a
Linking .pio\build\V10\firmware.elf
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE10showPixelsER15PixelControllerILS1_66ELi1ELm4294967295EE[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE10showPixelsER15PixelControllerILS1_66ELi1ELm4294967295EE]+0x20): undefined reference to `fl::I2S_Esp32::showPixels(unsigned char, PixelIterator&)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE11endShowLedsEPv[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE11endShowLedsEPv]+0x0): undefined reference to `fl::I2S_Esp32::endShowLeds()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o:(.literal._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE13beginShowLedsEi[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE13beginShowLedsEi]+0x0): undefined reference to `fl::I2S_Esp32::beginShowLeds(int, int)'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o: in function `fl::ClocklessController_I2S_Esp32_WS2812Base<(EOrder)66>::showPixels(PixelController<(EOrder)66, 1, 4294967295ul>&)':
C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/lib/FastLED-master/src/platforms/esp/32/clockless_i2s_esp32s3.h:80:(.text._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE10showPixelsER15PixelControllerILS1_66ELi1ELm4294967295EE[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE10showPixelsER15PixelControllerILS1_66ELi1ELm4294967295EE]+0x47): undefined reference to `fl::I2S_Esp32::showPixels(unsigned char, PixelIterator&)'  
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o: in function `fl::ClocklessController_I2S_Esp32_WS2812Base<(EOrder)66>::endShowLeds(void*)':
C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/lib/FastLED-master/src/platforms/esp/32/clockless_i2s_esp32s3.h:86:(.text._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE11endShowLedsEPv[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE11endShowLedsEPv]+0x9): undefined reference to `fl::I2S_Esp32::endShowLeds()'
C:/Users/markf/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/V10/src_x/V10/V10.cpp.o: in function `fl::ClocklessController_I2S_Esp32_WS2812Base<(EOrder)66>::beginShowLeds(int)':
C:\Users\markf\Documents\PlatformIO\Projects\BuddyCompass_VX/lib/FastLED-master/src/platforms/esp/32/clockless_i2s_esp32s3.h:73:(.text._ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE13beginShowLedsEi[_ZN2fl40ClocklessController_I2S_Esp32_WS2812BaseIL6EOrder66EE13beginShowLedsEi]+0xf): undefined reference to `fl::I2S_Esp32::beginShowLeds(int, int)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\V10\firmware.elf] Error 1
'''

I will now look into the spi diver, thanks for your suggestion

zackees, 2025-08-20:
Make sure you are on FastLED master. Very important for our spi driver for ESP32.

Yes, the spi driver can bind to different ports.

https://github.com/FastLED/FastLED/blob/48db45c8d9dde4c6ddc83205602256a03cc4dbf7/src/platforms/esp/32/spi_ws2812/strip_spi.h#L46

The default is auto detect, but it's not heavily tested. What I think it happening is that the radio binds to it's preferred SPI port and the auto detect engine for FastLED is detecting this and binding to a secondary port. This would explain why order of initialization matters. FastLED is flexible in the SPI it binds to, your radio is probably hardcoded for one or the other.

Keep in mind that you are on the frontier here. If you find a solution that works for you please merge a fix back to us, or at least let us know what your solution was.

I might also suggest you try forking this repo and build your app directly into dev/dev.ino. This will give you maximum access to the FastLED codebase.

If you do this keep in mind the following:
  * Run `./install` so you get the code completion json at the project root
  * Install the `clangd` extension if you want code completion. VSCode intellisense is disable in this repo because of reasons. Clangd is lightyears ahead, you'll love it.

Seitekful, 2025-08-20:
It worked! Thank you for your insight.

I changed `SPI_HOST_MODE_AUTO` to `SPI_HOST_MODE_3` and now the radio works on SPI2 and FastLED on SPI3.
https://github.com/FastLED/FastLED/blob/48db45c8d9dde4c6ddc83205602256a03cc4dbf7/src/platforms/esp/32/spi_ws2812/strip_spi.h#L46

Indeed maybe something about the auto-detection is not working right in my case. Since I'm using custom SPI pins I had to hardcode those into the radio-library (it was simply calling SPI.begin() before). So now there are "wrong" MOSI/MISO/SCK/NSS pin definition in pins_arduino.h. I dont now how the auto-detect works but maybe that has in influence.

Anyways, to recap, clockless spi is working on the ESP32-S2 using `#define FASTLED_ESP32_USE_CLOCKLESS_SPI` before `#include <FastLED.h>`. The only thing I had to do is to manually specify the SPI bus by modifying the library a bit. 

Now I would gladly contribute to his library but In not sure what to do in this case. It seems to me that it is working as intended but It could be useful to provide a way for specifying the SPI bus for the user? or leave it as is?

zackees, 2025-08-20:
Okay, i'll make the change in a branch and see if it just fixes it for you withotu the hack.

zackees, 2025-08-20:
Alright, fix is up. Point your FastLED dep to branch esp-spi

It should be this

```
[env:your_env]
lib_deps =
    https://github.com/FastLED/FastLED.git#esp-spi
```

zackees, 2025-08-21:
@Seitekful are you able to test the branch?

Don't be one of those users where I do a bunch of work to fix their issue then they ghost because their locally hacked copy of FastLED works.

zackees, 2025-08-21:
Don't worry about the work, it's just AI. Looking at the state, the AI actually had it right in a previous iteration then reverted back to internal and not global state tracking of spi devices.

I'll have another version up shortly at the same branch, then I'll ping you. I'd like to get this fix in for the pending 3.10.3 release if possible.

Seitekful, 2025-08-22:
Still having the same issue with esp-spi2 branch. Double checked and reverted back to my modified library: working as intended...