FastLED/FastLED #2067: Only first LED lights up 80% of the time

mcpacific, 2025-08-24:
With my WS2812B strips, after apply the latest FastLED, about 80% of the time using fill_solid only lights the first LED in the strips. I've tried this with two different strips and two different ESP8266 controllers.

For example, one strip has 50 leds. I setup the led array with:

`CRGB leds[NUM_LEDS];`

num_leds gets defined by the actual number from settings.

`fill_solid(leds, num_leds, CRGB::Red);`

Only the first LED turns red. The rest of the strip remains off.

4wheeljive, 2025-08-28:
I’m pretty new to this stuff, but I suspect that the “fix” of adding a second FastLED.show may be a coincidence, and that the problem involved something else. I also suspect that it was not a FastLED issue, but something in your hardware or software implementation.  

One thing I notice is that NUM_LEDS is capitalized when you declare CRGB leds, but it is lower case in your fill_solid function. I'm not sure exactly what you mean by 
> num_leds gets defined by the actual number from settings

but that's something you might want to doublecheck.

If the problem returns, I'd suggest asking for help on the reddit forum https://www.reddit.com/r/FastLED/ rather than here. And if you do, you should include your complete sketch (using https://gist.github.com or https://pastebin.com) as well as info about your hardware setup (power source, wiring, etc.). Good luck!

mcpacific, 2025-08-29:
@4wheeljive Thank you for your input. NUM_LEDS is a preprocessor macro defined as an integer constant. It is set as the maximum number of LEDS the sketch will allow. Later, in the global variable declaration section I define num_leds (int num_leds = NUM_LEDS;'.

In setup(), num_leds is redefined based on a value that was retrieved from a config.json file. This way I can use the same sketch for multiple ESP devices that have a different number of leds attached (in this case, an ESP8266 device).

There is no reason FastLED.show() should need to be repeated. The code worked perfectly without doing so until I updated to the latest version. I hope there will be no "if the problem returns" because I've added two FastLED.show()'s everywhere where one used to be, and now fill_solid() works as expected.

I'll check out the reddit forum!

Thanks again!

zackees, 2025-09-06:
Please post the entire sketch, including the platformio.ini file. Then I can debug this.

mcpacific, 2025-09-10:
@zackees Thank you for getting back to me. Here is my sketch. I'm using Arduino IDE (not PlatformIO) so I don't have a platformio.ini file.

`/*
 * Espalexa with RGB color devices. https://github.com/Aircoookie/Espalexa
 * Last Update: 8/28/2025
 * Using Espalexa with 433 mhz blinds instead of RGB LED strip.
 * WiFi Manager used to configure WiFi. LED will blink when WiFi is in config mode.
 * Press and Hold FLASH/BOOT > 3 seconds will start WiFi Config.
 * LED will blink when in config mode
 */

#include <WiFiManager.h>
#include <ArduinoJson.h>
#include <WiFiClient.h>
#include <Espalexa.h>
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
#include <ArduinoOTA.h>
#include <Ticker.h>
#ifdef ESP32
#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#else
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#endif

bool enableDebug = false;

#define NUM_LEDS 75 // Maximum number of LEDs
#define BRIGHTNESS 255
#define DATA_PIN 4
// #define CLOCK_PIN 22
#define TRIGGER_PIN 0 // Flash button on ESP8266
#ifndef LED_BUILTIN
#define LED_BUILTIN 2 // ESP32 DOES NOT DEFINE LED_BUILTIN
#endif

int LED = LED_BUILTIN;
char device_name[48];
char qty_leds[3];
int num_leds = NUM_LEDS;

bool shouldSaveConfig = false; // flag for saving data

#ifdef ESP32
#define ESPALEXA_ASYNC
const int buttonStateNotPressed = LOW;
const int buttonStatePressed = HIGH;
#else
const int buttonStateNotPressed = HIGH;
const int buttonStatePressed = LOW;
#endif
int lastButtonState = buttonStateNotPressed; // the current reading from the input pin
unsigned long lastDebounceTime = 0;          // the last time the output pin was toggled
unsigned long debounceDelay = 50;            // the debounce time in ms; increase if the output flickers
unsigned long resetDelay = 3000;             // the time required to press button to reset, in ms
bool rainbowActive = false;
bool rainbowBootActive = false;
unsigned long rainbowBootStart = 0;
uint8_t lastBrightness = 0;
uint32_t lastRgb = 0;
unsigned long directColorOnDelay = 400;

[... 597 lines cut ...]
        }
        else if (cmd == "status")
        {
            if (WiFi.isConnected())
            {
                telnetClient.println("\nWiFi Connected");
                telnetClient.println((String("SSID: ") + WiFi.SSID()).c_str());
                telnetClient.println((String("IP: ") + WiFi.localIP().toString()).c_str());
                checkSignalStrength();
            }
            else
            {
                telnetClient.println("\nWiFi Not Connected");
            }
            checkLedStatus();
            telnetClient.printf("Debug: %s\r\n", (enableDebug ? "Enabled" : "Disabled"));
        }
        else if (cmd == "reboot")
        {
            telnetClient.println("Rebooting device now...");
            ESP.restart();
        }
    }
}
`

zackees, 2025-09-19:
The new WS2812B chipset has a reset time of 280 uS, which is significantly longer than the previous spec of 50 uS. Out of an abundance of caution i'm going to increase the reset time. It's backwards compatible with the WS2812 timings.

closed 2025-09-19 by zackees

mcpacific, 2025-09-19:
@zackees It seems to work for the most part. However, my "rainbow mode" does not (running the Pride 2015 animation no longer works when I have Alexa change the color to lavender).

zackees, 2025-09-19:
prove to me this isn't user error for this one.