FastLED/FastLED #2068: Panic: rmt: rmt_tx_enable(765): channel not in init state

razvanphp, 2025-09-06:
Not sure what is causing this, as it does not happen all the time, just after boot or on transitions (pulsing leds).

```
Starting transition task
11:09:43.714 > 
11:09:43.890 > E (19708) rmt: rmt_tx_enable(765): channel not in init state
11:09:43.906 > E (19708) led_strip_rmt: led_strip_rmt_refresh_async(105): enable RMT channel[ failed
11:09:43.906 > 
11:09:43.954 > abort() was called at PC 0x4037df7b on core 0
  #0  0x4037df7b in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:49

11:09:43.954 > 
11:09:43.954 > 
11:09:43.954 > Backtrace: 0x4037685d:0x3fce62a0 0x4037df85:0x3fce62c0 0x403849e1:0x3fce62e0 0x4037df7b:0x3fce6360 0x420044fb:0x3fce6390 0x42148f4b:0x3fce63b0 0x4205c8aa:0x3fce63d0 0x420034e3:0x3fce63f0 0x4205ceea:0x3fce6420 0x4205d7e5:0x3fce6480 0x4037ebc9:0x3fce64f0
  #0  0x4037685d in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:469
  #1  0x4037df85 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:87
  #2  0x403849e1 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/abort.c:38
  #3  0x4037df7b in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:49
  #4  0x420044fb in (anonymous namespace)::RmtStrip::drawAsync() at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:108 (discriminator 1)

11:09:44.311 > 
11:09:44.311 > 
11:09:44.311 > 
11:09:44.311 > 
11:09:44.311 > ELF file SHA256: 8b1312563
11:09:44.311 > 
11:09:44.311 > Rebooting...
11:09:44.311 > ESP-ROM:esp32s3-20210327
11:09:44.311 > Build:Mar 27 2021
11:09:44.311 > rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
11:09:44.311 > Saved PC:0x403766a1
  #0  0x403766a1 in esp_restart_noos at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:162

11:09:44.311 > SPIWP:0xee
```

also sometimes this one, not sure if related:

```
6930 0x4037db2d:0x3fce6950 0x40384861:0x3fce6970 0x4037db23:0x3fce69f0 0x42004537:0x3fce6a20 0x42148f87:0x3fce6a40 0x4205c55e:0x3fce6a60 0x4200351f:0x3fce6a80 0x4205cbc6:0x3fce6ab0 0x4205d4c1:0x3fce6b10 0x4037ea21:0x3fce6b80
  #0  0x4037db2d in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:87
  #1  0x40384861 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/abort.c:38
  #2  0x4037db23 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:49
  #3  0x42004537 in (anonymous namespace)::RmtStrip::drawAsync() at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:108 (discriminator 1)
  #4  0x42148f87 in RmtController5::showPixels() at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/platforms/esp/32/rmt_5/idf5_rmt.cpp:72
  #5  0x4205c55e in ClocklessController<4, 250, 625, 375, (EOrder)66, 0, false, 5>::endShowLeds(void*) at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/platforms/esp/32/rmt_5/idf5_clockless_rmt_esp32.h:52
  #6  0x4200351f in CFastLED::show(unsigned char) at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/FastLED.cpp:157
  #7  0x4205cbc6 in CFastLED::show() at .pio/libdeps/esp32-s3-devkitc-1/FastLED/src/FastLED.h:796
      (inlined by) transition(CRGBW, CRGBW, int, int, int, int, LedManager*, int, int) at src/Led/LedManager.cpp:405
  #8  0x4205d4c1 in transitionLedTask(void*) at src/Led/LedManager.cpp:209
  #9  0x4037ea21 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

02:43:57.614 > 
02:43:57.614 > 
02:43:57.614 > 
02:43:57.614 > 
02:43:57.614 > ELF file SHA256: 5aaf673f7
02:43:57.614 > 
02:43:57.614 > E (11463) esp_core_dump_flash: Not enough space to save core dump!
02:43:57.614 > E (11469) esp_core_dump_elf: Failed to prepare core dump storage (257)!
02:44:00.261 > E (11475) esp_core_dump_common: Core dump write failed with error=257
02:44:00.261 > Rebooting...
02:44:00.261 > �ESP-ROM:esp32s3-20210327
```

Relevant code:
```C++
CRGB toCRGB(CRGBW rgbw)
{
    return CRGB(rgbw.r, rgbw.g, rgbw.b);
}

void transition(CRGBW currentColor, CRGBW newColor,
                int start_strip, int end_strip, int start_board, int end_board,
                LedManager *ledManager, int iterations, int delay)
{

    for (size_t k = 0; k <= iterations; k++)
    {
        CRGBW tmpColor = linearBlend(currentColor, newColor, k / 60.0);
        for (size_t i = start_strip; i <= end_strip; i++)
        {
            ledManager->LedStripLeds[i] = toCRGB(tmpColor);
        }
        for (size_t i = start_board; i <= end_board; i++)
        {
            ledManager->LedBoardLeds[i] = toCRGB(tmpColor);
        }
        FastLED.show();
        vTaskDelay(delay / portTICK_PERIOD_MS);
    }
    
    Serial.println("Led transition done");
}
```

FastLED @ 3.10.1 @ platform.io

Any idea how to investigate this further? This started relatively recently, after upgrading to Arduino Core v3.3.0, was on v2.x before.

Thank you!
R

zackees, 2025-09-07:
You have one strip or two strips?

What's your board?

This should work with one or two strips on all the major esp32 versions except c2.

I2S has serious issues. It makes it very hard to reflash the device (you have to hold down reset and then release it to force the upload), additionally, stdout becomes disabled. There's a newer I2S driver in the works that fixes all of this and I hope to make the I2S driver the default. Only use I2S if you are pushing a massive number of LEDS.

I suggest you try out the SPI driver for WS2812 (yes, it works with just one pin, the clock is ignored).

[image]

razvanphp, 2025-09-07:
So just with `FASTLED_RMT5=0` I got this Panic:

```
17:01:58.036 > SPIWP:0xee
17:01:58.036 > mode:DIO, clock div:1
17:01:58.036 > load:0x3fce2820,len:0x116c
17:01:58.036 > load:0x403c8700,len:0xc2c
17:01:58.036 > load:0x403cb700,len:0x3108
17:01:58.036 > entry 0x403c88b8
17:01:58.036 > E (390) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver
17:01:58.036 > 
17:01:58.036 > abort() was called at PC 0x420c1b0b on core 0
  #0  0x420c1b0b in check_rmt_legacy_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/deprecated/rmt_legacy.c:1413

17:01:58.036 > 
17:01:58.036 > 
17:01:58.036 > Backtrace: 0x4037e289:0x3fceb200 0x4037e251:0x3fceb220 0x40384f85:0x3fceb240 0x420c1b0b:0x3fceb2c0 0x42089eba:0x3fceb2e0 0x40376066:0x3fceb310 0x403ccbf1:0x3fceb340 0x403ccfad:0x3fceb380 0x403c8925:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x4037e289 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:469
  #1  0x4037e251 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:87
  #2  0x40384f85 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/abort.c:38
  #3  0x420c1b0b in check_rmt_legacy_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/deprecated/rmt_legacy.c:1413
  #4  0x42089eba in do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:104
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:207
  #5  0x40376066 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:935 (discriminator 1)

17:01:58.036 > 
17:01:58.036 > 
17:01:58.036 > 
17:01:58.036 > 
17:01:58.036 > ELF file SHA256: f5ebccae3
```

Not even adding `FASTLED_RMT_BUILTIN_DRIVER=1` does not help. Using Arduino Core v3.3.0.

Just want to mention that the setup that worked before was latest platform.io Arduino Core for esp32 v2.0.17 (based on IDF v4.4.7) + FastLED 3.10.1 (found it in my old logs), you mean this combo works with RMT4?

I know the SPI way, but I don't have any SPIs left on my MCU 🙁 all are used already. Also, I prefer RMT, as bit banging will block the CPU cycles, is RMT4 still hardware, or?

I will try to port my LED code to the template and invite you.

Cheers,
R

zackees, 2025-09-07:
The error you are getting is because fast led is doing a partial compile for RMT4 and RMT5

espressif made it a mandate that if both drivers are linked in at the same time then it aborts on boot up.

hence the reason why we had to migrate to RMT5. for example some of the boards have Ws2812 status lights that use wes2812. This will blow up the prior.

zackees, 2025-09-07:
I'm refactoring rmt and adding worker pools.

I'm traveling so i can test with a physical device, but if these tests pass you might want to try forking this PR and see if it resolves your issue.

https://github.com/FastLED/FastLED/pull/2072

closed 2025-09-07 by zackees