FastLED/FastLED #2088: How to add standalone prefix to each output of pixel data

second-string, 2025-09-22:
I'd like to implement the UCS7604 protocol in FastLED. Unfortunately, the manufacturers don't display any protocol info their datasheets. I have UCS2904 and UCS7604 LEDs, and a closed-source driver than can drive both. The UCS2904 can be driven with FastLED using the TM1804 protocol, as they have similar (probably identical timings).

The UCS7604 one has two major differences from the hardware protocol perspective:
1. There is a two-chunk preamble that sets things like max current per channel, speed of data section (800k vs. 1.6M), etc. First chunk is 8 bytes, second is 7, they are separated from each other and the main color data by ~260uS.
2. There are 16 bits per R, G, B, and W channel instead of 8 (this should be changeable according to the data sheet, but I have to reverse engineer the proper bits to change in the preamble)

Difference #2 is not a huge deal for me right now, I can scale the incoming 1-byte data into 2-byte data in my firmware. Difference #1 is what I need to patch into FastLED. I'm not sure the best place to do this, however. I don't think I can just shove an extra few bytes into the front of the  color data frame before transmitting to act as the config bytes because of the 260uS spacing. I will give it a try though, in case the IC doesn't actually care about the spacing and it's just counting bytes.

The clockless implementation I'm currently using is an ARM chip, which has its output written in the assembly loops. I think I would have to adapt the ASM loop in the m0clockless.h file to add two more loops before the color data, with an appropriate delay in between them. Any other ideas as an easier first way to implement this as a proof of concept?

Here are some labelled screenshots of the output for UCS7604, also with the Saleae capture file attached (Saleae's Logic app is free to download if you want to look through it). I've been able to figure out a couple of the config bits by toggling my driver's settings, but it doesn't have a setting for data width. My guess is that it's in the first two nibbles of the second preamble chunk, but I can't test it until I have a base implementation working from FastLED.

Not looking for anyone to do the work for me, just looking for guidance on the best architecture/implementation by those more familiar with the repo

[image]

[image]

[image]

[UCS7604 Red RGBW.sal.zip](https://github.com/user-attachments/files/22466349/UCS7604.Red.RGBW.sal.zip)

second-string, 2025-09-22:
RGBW was one that I was poking around it, I'll go a little deeper and check it out.

The bitbang wrapper solution you've mentioned was one of the first things I was thinking of, but I wasn't sure if I could bitbang fast enough and tight enough to adhere to the timing requirements. The RGBWEmulated controller just has to do pre-processing on the data, then can shove it in the normal implementation-specific `show` function to blast it out as efficiently as possible.

Do you know off the top of your head if that's possible (or if it has been done in the repo  already) for port/pin register-based control? I'll give it a try regardless if the non-spaced hackier version doesn't work, but will take any prior knowledge I can get.

zackees, 2025-09-27:
Checking in: how's it going?

zackees, 2025-10-07:
See the readme for instructions on how to run this chipset. Let me know if i got it wrong as literally can't test this.

second-string, 2025-10-07:
Left a few comments even though you didn't ask for them. I'll test the merged implementation with the delays in the next couple of weeks

zackees, 2025-10-08:
we did you leave the comments?

second-string, 2025-10-29:
I've tried a couple times to integrate the FastLED code into our repo but it has become unmanageable in the last ~year since we first integrated it. Previously, I simply dropped the contents of the `src/` file into it's own CMake directory in our repo and did a bit of work stubbing or wrapping a couple things. Now with the whole overriding of the C std library files, etc, it's much more complicated. Also, everything is chaotic and in different places (see all the random files in the top level of the repo alone).

It seems the AI use is too heavy, as all of these tens (hundreds?!) of daily commits have different changes with zero explanation in the commit message. Maybe someone else will pick this up to test it in the future but I am not able to.

uzlonewolf, 2025-11-11:
Thinking about it a bit more, to increase reliability we should probably increase the low times a bit to make sure they are not being run right at the edge.  Adjusting them so they run at the spec'd 800 kbps would be:

Low Speed:
* T0H: 420ns
* T0L: 830ns
* T1H: 840ns
* T1L: 410ns

High Speed:
* T0H: 210ns
* T0L: 415ns
* T1H: 420ns
* T1L: 205ns

zackees, 2025-11-11:
All modes now supported. Give it a whirl and let me know if it works for you or if there are any problems